wiedehopf/tar1090 #189: Documentation of the protocol

gvanem, 2022-06-25:
Hello folks!

I'm trying to adapt the  `html` root of this project into my own [Dump1090](https://github.com/gvanem/Dump1090) Windows-version which uses the excellent
[Mongoose](https://github.com/cesanta/mongoose) library for HTTP and other network stuff.

My program works just fine with the original `web_root` of Salvatore Sanfilippo original work.

So I have some problems understanding the json-format of the dynamic content used in Tar1090 (and others like FlightAware?).

E.g. in response to a `GET /data/aircraft.json?...`, my program sends this: 
```
{ "now:":18446744072021792.0, "messages": 4, 
  "aircraft": [
     {"hex":"484165", "flight":"", "lat":59.766742, "lon":4.908142, "altitude":17350, "track":356, "speed":422},
     {"hex":"478095", "flight":"WIF677", "lat":60.317093, "lon":5.210171, "altitude":550, "track":170, "speed":110},
     {"hex":"4783CC", "flight":"WIF71A", "lat":60.369361, "lon":5.248424, "altitude":2050, "track":204, "speed":152},
     {"hex":"47A3B0", "flight":"", "lat":60.495850, "lon":5.201648, "altitude":4175, "track":217, "speed":210} ]
}
```

Edited for clarity. I think this is valid Json. But I have probably missed some fields. Which are expected and needed?

wiedehopf, 2022-06-25:
https://github.com/wiedehopf/readsb/blob/dev/README-json.md

dump1090-fa changed the field names.

tar1090 should be able to handle the "old" field names i think ... not sure though. (well with dump1090-mutability)
That "now" is definitely wrong, it should be in seconds and you have nanoseconds or something :)
And i believe even dump1090-mutability already had the fields seen and seen_pos, add those fields and it might start working.
Otherwise change to the field names as described in the readme i linked, you don't need all of them as some of them sometimes don't apply.

You could always try porting readsb to windows.
But that's probably a bit annoying as i've added a lot of libraries it uses.

gvanem, 2022-06-25:
My response to a `/data/receiver.json?_=1656176836005` is:
```json
{ "version": "0.1", "refresh": 1000, "history": 0, "lat" : 60.301682, "lon" : 5.320877 }
```

which shows the map correctly with my home-pos.

> That "now" is definitely wrong,

Yes it was. Now the `/data/aircraft.json` looks like (2 planes):
```
{"now:": 1656178720, "messages": 2, "aircraft" : [
  {"hex": "48AE00", "flight": "", "lat": 59.330408, "lon": 4.432415, "altitude": 37975, 
   "track": 313, "speed": 518, "type": "adsb_icao", "seen":2, "seen_pos": 2},
  {"hex": "47C1F7", "flight": "FOX4786", "lat": 60.255122, "lon": 5.231258, "altitude": 1775, 
   "track": 168, "speed": 172, "type": "adsb_icao", "seen": 2, "seen_pos": 2}
  ]
}
```

Does it look OK?  But I'm just faking the `"seen": 2` and `"seen_pos": 2` for now.
But even with this Json, the `Tar1090/index.html` and *Debug Shall All* enabled shows nothing. 
What should I look for? Is it possible to debug your stuff with Node?

> You could always try porting readsb to windows.

Never heard of it before. But I'll try to built it for Windows.

gvanem, 2022-06-25:
Just discover my program received position from a plane 700 km away! (callsign FLI459). Excellent radio conditions now it seems.

wiedehopf, 2022-06-25:
have you checked that it's loading aircraft.json via the network tab in F12?

Add the messages field to aircraft ... maybe rssi as well? Thought it should work without rssi.
Check the console log ... F12

It's just javascript, i don't use nodejs. (well i use npm to create the customized version of openlayers but for nothing else).

wiedehopf, 2022-06-25:
Really you should at least base your stuff on dump1090-mutability .... as it has much better demodulation code.
Which is further improved in dump1090-fa / readsb (which performs similar).

Honestly i'd just recommend running a minimal Debian / Ubuntu in Virtualbox and use this software stack: https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

You can still use dump1090-mutability for windows to connect to the SDR and then use the data in the VM to avoid MLAT issues when running the decoder in the VM.

gvanem, 2022-06-26:
> have you checked that it's loading aircraft.json via the network tab in F12?

Yes, I see the `aircraft.json` just fine in Chrome's Dev-Console.

> Really you should at least base your stuff on dump1090-mutability 

You mean this https://github.com/adsbxchange/dump1090-mutability that has not been updated in 5 years?
But I'll try to make my web-root similar to the others.

wiedehopf, 2022-06-26:
Well i don't know what you're basing it on, it seems to be even older. Does your codebase even have the much improved 2.4 MHz demodulator? https://github.com/mutability/dump1090/blob/archiver/demod_2400.c
I know for a fact that i've checked that tar1090 works with dump1090-mutability.
(because it's a debian package and was more common when i starded working on tar1090)

Have you tried adding the messages field to each aircraft?
Again i think it might be required.

The console log (F12) can help you find js Exceptions.

wiedehopf, 2022-06-26:
Did a test, this json works just fine.
```json
{ "now" : 1656219853.808,
  "messages" : 10935,
  "aircraft" : [
{"hex":"44a831","flight":"JAF4T   ","altitude":37000,"speed":458.5,"track":147.11,"lat":48.822830,"lon":6.711661,"seen_pos":0.815,"seen":0.3},
{"hex":"44ce72","flight":"BEL31M  ","altitude":39025,"speed":435.8,"track":148.61,"lat":48.648514,"lon":6.914767,"seen_pos":1.734,"seen":0.3},
```

No clue what you're doing wrong.
now actually needs to be updating btw .... 
just in case you're using something static

wiedehopf, 2022-06-26:
Tested even more minimal:
```json
{ "now" : 1656220248.038,
  "messages" : 133708,
  "aircraft" : [
{"hex":"a2a2df","flight":"UAL106  ","altitude":39000,"speed":535.5,"track":92.57,"lat":48.492582,"lon":6.121192},
{"hex":"44d2ab","flight":"JAF2323 ","altitude":37000,"speed":452.1,"track":146.73,"lat":48.964249,"lon":6.548936},
```

Are you updating now, it's required to be incrementing otherwise aircraft are just gonna stand still and never updated.
They should still show.

Are they showing in the table on the right?
Try disabling webGL in the settings, maybe your browser is being stupid with displaying the icons.

Anyhow you can also add console.log to for example updateData in planeObject.js, see if that's ever called.
You can check if processAircraft in script.js is ever called.

Just check if PlanesOrdered / PlanesOrdered[0] exist on the js console.

gvanem, 2022-07-02:
So after adding a `console.log ("Got Ajax data: ", data);` to the Ajax .done() handler, I see this in Chrome's console:
[image]

Looks OK to to me so I've no idea where the processing of this Ajax data stops.

So what is the relationship between the Json I send and this Ajax-data that's received? I'd assume it's a 1-to-1 relationship.
But it doesn't look so.

wiedehopf, 2022-07-02:
Follow the function call stack.

You have a part in the code where the data is present.
You have a part in the code where nothing is happening.

Investigate what happens on the way.
In other words, debug the code.

I might know what's happening, but i think it'll be a better experience for you if you find the exact line where the issue happens.

wiedehopf, 2022-07-02:
you haven't for example examined what happens in updateData ....
does it get the data from the ajax call?

what does it take to get to the for loop from there?

gvanem, 2022-07-02:
It was a small typo on my part; a:
* `"now:" 1656772434.202,..`. Instead of
* `"now" 1656772434.202,..`

So now it seems to work:
[image]

closed 2022-07-02 by gvanem

wiedehopf, 2022-07-02:
you can btw also set breakpoints in the browser and look at variable values.

wiedehopf, 2022-07-02:
yeah i just saw it in the screenshot.

too bad you started the windows porting with such an old codebase :/

Good luck with your project.

gvanem, 2022-07-02:
> too bad you started the windows porting with such an old codebase :/

I tried your readsb, but it's way so Unix-centric. Besides the new codebases like from FA, is a mess.
I've added SDRPlay support too, but it's not working quite good yet.

BTW. How do I get the plane's icon. Now all I see is a mess of tracks not knowing where the plane really is.
[image]

gvanem, 2022-07-02:
Ah. Found it. Enabling **Airpos Web GL** shows the icon.

wiedehopf, 2022-07-02:
That is enabled by default.

Well i'm not saying porting it wouldn't be some work but it's possible.
Anyhow, you could always add the 2.4 MHz demodulator, the decoding for more fields.