wiedehopf/tar1090 #190: tar1090 web interface not reachable - how to change localhost address?

kinteki, 2022-06-26:
Hi,

I have a new installation of tar1090, but not reachable the webinterface. How can I change the IP address?
On my Rpi4 previously installed:  Birdnet. 

some sys info:

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Hardware        : BCM2835
Revision        : c03115
Serial          : 1000000027999f56
Model           : Raspberry Pi 4 Model B Rev 1.5

zoli@rpi4:~/BirdNET-Pi$ systemctl is-active tar1090.service
active

Sincerely, 
Zoltan

wiedehopf, 2022-06-26:
You need a decoder as well: https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

You also need to install lighttpd if you don't have that.

wiedehopf, 2022-06-26:
Asking how to change the IP address has nothing to do with tar1090.
I'd recomend just using DHCP .... so the IP is gonna be whatever your router assigns the device.
You are connected to the device, so you know the IP, so what is the issue?

kinteki, 2022-06-26:
I have decoder, and antenna. and installed this: [Automatic installation for readsb](https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb). when installed alone its works properly. But I installed Birdnet, either.

My device IP address is handled: 192.168.1.157.  My problem is that when I try to reach 192.168.1.157/tar1090 i cant reach the webserver.  FR24feeder works properly.
[image]

Somehow the Birdnet's webserver' chunk appeared.

Can I change the address the web server?

wiedehopf, 2022-06-26:
Change the port used by lighttpd.
I'm sure you can google how to do that / which config file is being used on Raspbian.

closed 2022-06-26 by wiedehopf