wiedehopf/tar1090 #192: Odd Cross-Map Tracking Lines

Tarsi210, 2022-07-08:
When having TAR1090 open for long periods of time, the display will sometimes devolve into having these perpetual cross-map track lines that do not correspond to actual flight tracks.   I'm unsure of why but wanted to report it.   Does not matter on the browser and if I open another browser at the time this appears, the lines persist, although not as many (some resolve).   It's all very strange.  
[image]

wiedehopf, 2022-07-08:
can you show /tar1090/?ptracks ?
are the lines there as well?

i've never seen this but i don't leave it open for long.

Tarsi210, 2022-07-08:
Nope, once I go to pTracks the tracks start over from none and start to build fresh, as if they have no history.   In fact, loading pTracks caused the main view to reset all track history, removing the problem (temporarily).    I have my display on my large 70" monitor in my office and view my planes all day, so that's why I notice this.    I'm on the latest build.

wiedehopf, 2022-07-08:
if you don't even have ptracks, your webserver configuration is probably bad.

the dots beyond the screen might be issues with either bad decodes or some data loop .... not certain.
i don't know which underlying decoder you run.

Tarsi210, 2022-07-08:
Yeah, I really don't know what's going on.    So, tar1090 is running on this server and using the JSON feeds from the adsbexchange feeder as my dump1090-fa is actually running over on my Pi but I'm running tar1090 on a "real" server.    Using lighttpd as the webserver.    This might be an adsbexchange feed problem, then.    Thanks.    I'll poke at it.

Is there any way to set tar1090 to use a remote URL for both 1090 and 978 aircraft.json files instead of having to do this hack with adsbx?

wiedehopf, 2022-07-09:
Well update the feed client then.

Why would you not use the feed client, it's the best!
Might actually be a bug i had over a year ago in readsb.

closed 2022-08-12 by wiedehopf