wiedehopf/tar1090 #198: [ERROR] Error in line 226 when executing: bash tar1090-install.sh /run/readsb

MDQ0512, 2022-08-04:
I can't use a one-click installation link.
when i installing tar1090，It prompts an error.

Here is the error message：
967 (process ID) old priority 10, new priority 10
HEAD is now at 2876b1f database update (to keep the repository small, this commit is replaced regularly)
HEAD is now at cdcf5d8 update install.sh.
sed: can't read /usr/local/share/tar1090/tmp.gHIpeBipxm/index.html: No such file or directory
[ERROR] Error in line 266 when executing: sed -i -e "s/let databaseFolder = .*;/let databaseFolder = \"db-$DB_VERSION\";/" "$TMP/index.html"
[ERROR] Error in line 226 when executing: bash tar1090-install.sh /run/readsb

wiedehopf, 2022-08-04:
that can't really happen.

if it does ... would seem like you have a broken filesystem.

```
sudo rm  /usr/local/share/tar1090  -rf
```
try removing and running the install again.

MDQ0512, 2022-08-04:
The raspberry OS I use is brand new. I started installing tar1090 when the OS was installed. I have rewritten the system four times, but it is the same every time. And after my investigation, the problem should appear in the installation of readsb, because the installation process of readsb is very short compared with a month ago.

wiedehopf, 2022-08-04:
Could be a broken sd-card. As in completely malfunctioning.

wiedehopf, 2022-08-04:
What is the error now?

MDQ0512, 2022-08-04:
This is my installation process：
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"
--2022-08-05 01:16:24--  https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh
Resolving github.com (github.com)... 140.82.112.3
Connecting to github.com (github.com)|140.82.112.3|:443... failed: Connection timed out.
Retrying.

--2022-08-05 01:18:49--  (try: 2)  https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh
Connecting to github.com (github.com)|140.82.112.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/readsb-install.sh [following]
--2022-08-05 01:18:51--  https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/readsb-install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8852 (8.6K) [text/plain]
Saving to: ‘STDOUT’

-                               100%[=====================================================>]   8.64K  --.-KB/s    in 0.005s  

2022-08-05 01:18:52 (1.87 MB/s) - written to stdout [8852/8852]

651 (process ID) old priority 0, new priority 10
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libncurses-dev' instead of 'ncurses-dev'
gcc is already the newest version (4:8.3.0-1+rpi2).
git is already the newest version (1:2.20.1-2+deb10u3).
libncurses-dev is already the newest version (6.1+20181013-2+deb10u2).
libusb-1.0-0-dev is already the newest version (2:1.0.22-2).
libzstd-dev is already the newest version (1.3.8+dfsg-3+rpi1+deb10u2).
libzstd1 is already the newest version (1.3.8+dfsg-3+rpi1+deb10u2).
lighttpd is already the newest version (1.4.53-4+deb10u2).
make is already the newest version (4.2.1-1.2).
ncurses-bin is already the newest version (6.1+20181013-2+deb10u2).
zlib1g is already the newest version (1:1.2.11.dfsg-1+deb10u1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1+deb10u1).
librtlsdr-dev is already the newest version (0.6-1+rpt1).
librtlsdr0 is already the newest version (0.6-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 143 not upgraded.
remote: Enumerating objects: 98, done.
remote: Counting objects: 100% (98/98), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 58 (delta 45), reused 16 (delta 7), pack-reused 0
Unpacking objects: 100% (58/58), done.
From https://gitee.com/smallmeng/readsb
 * branch            stale      -> FETCH_HEAD
 + 1cb85ad...739bc3e stale      -> origin/stale  (forced update)
HEAD is now at 739bc3e receiverCount mods
rm -f *.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o readsb viewadsb cprtests crctests convert_benchmark
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c argp.c -o argp.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c anet.c -o anet.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c interactive.c -o interactive.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c mode_ac.c -o mode_ac.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c mode_s.c -o mode_s.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 739bc3e (committed: Wed Aug 3 20:01:31 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR -I/ -g -O3 -march=native -c comm_b.c -o comm_b.o
[... cut ...]
967 (process ID) old priority 10, new priority 10
HEAD is now at 2876b1f database update (to keep the repository small, this commit is replaced regularly)
HEAD is now at cdcf5d8 update install.sh.
sed: can't read /usr/local/share/tar1090/tmp.gHIpeBipxm/index.html: No such file or directory
[ERROR] Error in line 266 when executing: sed -i -e "s/let databaseFolder = .*;/let databaseFolder = \"db-$DB_VERSION\";/" "$TMP/index.html"
[ERROR] Error in line 226 when executing: bash tar1090-install.sh /run/readsb
root@raspberrypi:/home/pi#

MDQ0512, 2022-08-04:
This card is in good condition because I was using it not too long ago

wiedehopf, 2022-08-04:
Please try installing tar1090 separately, thank you:
 https://github.com/wiedehopf/tar1090#installation

wiedehopf, 2022-08-04:
Why are you using this repository: From https://gitee.com/smallmeng/readsb

It seems to make random changes that don't actually do anything.

wiedehopf, 2022-08-04:
Are you getting the readsb install script replaced somehow ..... before it's executed?
That URL is not part of my readsb install script.

MDQ0512, 2022-08-04:
My country can't connect to github very well, so I fork these files to our country's git repository

MDQ0512, 2022-08-04:
But the content of the source code of readsb is the same

wiedehopf, 2022-08-04:
why do you put your name onto commits and give it new commit descriptions?
anyhow that's fine i guess and irrelevant.

wiedehopf, 2022-08-04:
Anyhow please run the changed tar1090 install script and then check:

```
ls /usr/local/share/tar1090/.instance_tmp
```
This is the directory where it copies the html stuff to, see this line in the script (and what might not be working for whatever reaons, i suspect the git clone befroe doesn't work?)
```
266     cp -r -T html "$TMP"                                                                                                                                                                   
```

MDQ0512, 2022-08-04:
Because we have a lot of aviation enthusiasts who are using your install script, but not many people know English, and because they can't use github, let me help them

wiedehopf, 2022-08-04:
Yeah i'm just confused why it has custom commits instead of the repository being a one to one copy.
The commits are also very strange ....

MDQ0512, 2022-08-04:
Sorry if I committed changes to your repository, please forgive me, I may have clicked wrong by accident, I just wanted to sync the files

wiedehopf, 2022-08-04:
It's ok ... just confusing.

MDQ0512, 2022-08-04:
Do you have any ideas for developing other languages?

wiedehopf, 2022-08-04:
I have no plans to support other languages, that's a lot of work.

Did you figure out what the install error was? It just couldn't git clone the files yes?

MDQ0512, 2022-08-04:
It's not clear to me what the problem was, but the problem is solved for now.

MDQ0512, 2022-08-04:
Thank you very much for your help, at the same time you have a way to add other languages, you can contact me, I have completed the translation for Chinese

MDQ0512, 2022-08-04:
Thank you very much for your help, and if you have ideas for adding other languages, you can contact me, I have completed the translation for Chinese

wiedehopf, 2022-08-04:
Of the tar1090 webinterface? That's very nice.
Just provide it as a fork.
I suppose i can start adding some interface that can display a different language.
Something like:
```
translate("Groundspeed")
```
Then some function which for English doesn't do anything.
For another language it would replace the word.
Redoing the code to support that will be a bit tedious.
And also the html stuff would all need to be generated by the javascript (which is still a target of mine).

I'll get back to you if i can ever provide such a utility.

MDQ0512, 2022-08-04:
Okay, that's great, looking forward to your good news

closed 2022-08-05 by wiedehopf