wiedehopf/tar1090 #199: Default sorting

EDXWer, 2022-08-10:
Is it possible to set the default sorting (distance, flight level) by URL?

wiedehopf, 2022-08-10:
no.

JRG1956, 2022-08-11:
Default column sort order seems to have changed recently.

With my tar1090 version installed 28 July, the data table defaults to being sorted lowest to highest on distance. With my version installed  10 August, the data table defaults to "random" - looking at the fully expanded data table I can't see that it is sorted on any of columns.

I'm not sure when the change started - but it has been in versions I've installed over at least the last week. Could it have something to do with the changes you made related to the max distance column when using /?pTracks  ?

closed 2022-08-12 by wiedehopf

wiedehopf, 2022-08-12:
@EDXWer https://github.com/wiedehopf/tar1090/blob/master/README-query.md