wiedehopf/tar1090 #200: [FEATURE] Indicator or Preferential of UAT over ADS-B

Tarsi210, 2022-08-28:
It seems (to me at least) that when an aircraft is transmitting both ADS-B and UAT and I am running tar1090 with both inputs consolidated, tar1090 will show the Source as ADS-B instead of UAT.   It would be nice to either be able to preference UAT display over ADS-B (if I choose) or to possibly somehow indicate that the aircraft is transmitting both?   Unless I'm missing something.  Thanks!

Squire298, 2022-08-28:
if you change the below to point to your "remote" then you will have UAT as the secondary and adsb as the primary.

I use combine1090 and point UAT to that.

# Change to yes to enable UAT/978 display in tar1090
ENABLE_978=no
# If running dump978-fa on another computer, modify the IP-address as appropriate.
# The URL needs to point at where you would normally find the skyview978 webinterface
URL_978="http://127.0.0.1/skyaware978"

my one.

# Change to yes to enable UAT/978 display in tar1090
ENABLE_978=yes
# If running dump978-fa on another computer, modify the IP-address as appropriate.
# The URL needs to point at where you would normally find the skyview978 webinterface
URL_978="http://192.168.1.206/combine1090"

this is the output. https://prnt.sc/2Xg28SzSlDnu

wiedehopf, 2022-08-28:
Squire that formatting is dreadful .......

Not sure i want to put in a switch for making UAT priority over ADS-B .... i'd recommend just using a separate interface for 978 if that matters to you.

wiedehopf, 2022-08-28:
Please set 
```
prefer978 = 30;
```

in config.js and see how that suits you.
If you lose 978 reception and still have 1090 reception, it will show the outdated 978 data for 30 seconds.
Or at least it should i can't really test.

wiedehopf, 2022-08-28:
Ok, pushed a better version.

prefer978 = 10;

That should be more than sufficient, only use higher values .. well i'm not sure why you would.

wiedehopf, 2022-08-28:
And if you like it, please make sure your data ends up at https://globe.adsbexchange.com/ as well :)

Just yesterday changed it so that the normal feed client will automatically also feed 978 if it's on the same machine (an extra feed client was necessary previously).

Hmm now i'm curious, you didn't mention initially how you put the 978 into tar1090, i assumed it was via ENABLE_978 in /etc/default/tar1090 and worked on that.
If you feed the 978 into readsb ... it's quite a different story. And quite honestly in that case you're out of luck because readsb will always use the best data source and 1090 is more reliable than 978 by quite a bit.

Tarsi210, 2022-08-30:
Thanks much, @wiedehopf !   And I already feed ADSB-X, have for some time. :)

The reason behind this request is not because I prefer 978 necessarily, but because my 978 antenna/system setup is newer than my 1090.   I have a relatively low amount of GA in my area so having a lot of data points to test my system with is difficult to get, to see what I am getting, if it's better or worse in different antenna/LNA configurations, etc.   You're right, most of the time I want 1090 but to tweak the system, I want to be able to artificially tweak it the other way, especially when it comes to turning on Persistent mode and watching for the reception patterns over a day/two days/etc.    Things like that.

So I very much appreciate it.   I've pulled this down and am seeing over the next few days what it looks like, thanks!

Tarsi210, 2022-08-30:
> Hmm now i'm curious, you didn't mention initially how you put the 978 into tar1090, i assumed it was via ENABLE_978 in /etc/default/tar1090 and worked on that. If you feed the 978 into readsb ... [...]

Yes, I used the ENABLE_978; I did change the 1090 over to readsb awhile ago (on your suggestion) and it works great, but the 978 worked out of the box coming through that interface so I left it.

wiedehopf, 2022-08-30:
If you're not using the separate 978 feed client (which has a separate interface for 978 traffic), be sure to update the normal feed scripts, they now also make sure to feed 978 data.