wiedehopf/tar1090 #203: White tar 1090 screen on mobile

Mifaxav, 2022-10-06:
Ive installed tar 1090 on a pi running piaware. On laptop, its totally fine! everything works. but on mobile, its the tar1090 interface but white, with no map, no results and a loading screen. It used to work before but i had to reinstall due to some other reasons. Any idea how this can be fixed?

wiedehopf, 2022-10-06:
It works fine on mobile, you're using the wrong IP or something.

Patty2k, 2022-10-06:
Me too, i use it allways on two receivers and one with combine1090 for hfdl ads-c too. Since commit a7ae8b462e80589e666360eefa73e8b21f1f3dc7y omky white Screen on mobile phone (Oppo find x2 pro, Firefox). On desktop or I switch the desktop mode in mobile Firefox, no Problems. Systems are all working fine. Undoing the "if (onMobile) {...." change in script.js did nothing. tried everything, but didn't really have the time to sift through all the changes.

wiedehopf, 2022-10-06:
i'll test.

closed 2022-10-06 by wiedehopf

Patty2k, 2022-10-06:
This was fast, work now. Thanks.

wiedehopf, 2022-10-06:
Well i broke it so i better fix it.

One user is user error, two users means i broke something :)

Mifaxav, 2022-10-07:
Dam, i post something a few hours before i sleep and its fixed as i wake up! Thanks @wiedehopf

Mifaxav, 2022-10-07:
Its still not working on mobile :(

Uninstalled and reinstalled using the sudo bash command

Mifaxav, 2022-10-07:
Nevermind. Ended up git cloning the repo and then 

Cd tar1090
Sudo bash install.sh

Worked for me! Thanks again @wiedehopf

Mifaxav, 2022-10-07:
Beep bop i am back. Tar 1090 works just fine but im still getting the white screen in 192.168.x.y/tar1090/?pTracks

Mifaxav, 2022-10-07:
Yeah i rlly dunno what's happening. Sometimes its fine sometimes it isnt