wiedehopf/tar1090 #212: Support coma seperated filter syntax in GUI

rickstaa, 2022-12-20:
I'm trying the filter more than one plane using the `Filter by ICAO hex id:` input field. When I use comma-separated syntax, however, nothing seems to happen (i.e. `aae2f1,a980cf`). I just read in [the documentation](https://github.com/wiedehopf/tar1090/blob/master/README-query.md) that you support this option through query variables (i.e. `https://globe.adsbexchange.com/?icao=aae2f1,a980cf`). Adding this feature to the GUI would significantly improve user experience since the page doesn't need to be reloaded.

wiedehopf, 2022-12-20:
That link selects them, doesn't really filter for them.
Just enable multi select and search for each hex.

The filters are js regex so you can just do aa11bb|bb11aa to filter for too hexes i believe .... might need braces.

rickstaa, 2022-12-20:
> That link selects them, doesn't really filter for them. Just enable multi select and search for each hex.
> 
> The filters are js regex so you can just do aa11bb|bb11aa to filter for too hexes i believe .... might need braces.

@wiedehopf Thank you for your answer and the chat support. For people looking at this in the future. I could select multiple planes by enabling the multi-select button `M` and then using the following syntax `151eee|151d51|7808c8|781335` in the `Filter by ICAO hex id:` field.

closed 2022-12-20 by rickstaa