wiedehopf/tar1090 #218: Query closest aircraft in json

SingingDwarf, 2023-01-30:
It would be great if tar1090 could show the closest aircraft within a json message, just as Virtual Radar Server does using the closestAircraft.json URL

This would mean that I could query the system and easily find the closest aircraft flying over my house, displaying on an eInk display or other system without pulling and manipulating the entire aircraft.json message.

closed 2023-01-31 by wiedehopf

SingingDwarf, 2023-01-31:
Although it is not clear from the reply above(?), it appears that this functionality may already have been implemented - https://github.com/wiedehopf/readsb/blob/dev/README-json.md

wiedehopf, 2023-02-01:
Why are you opening a new PR when i closed this one.

wiedehopf, 2023-02-01:
just read the readme

wiedehopf, 2023-02-01:
You found the relevant thing, the api port ... use it.

SingingDwarf, 2023-02-01:
I didn't understand why this one was closed initially with no comment, then figured it out from the JSON document (which wasn't easy to find, as it's not in the root - I stumbled across it via Google).

I then figured you misunderstood the request, as I don't think the API does what I want - show the single most closest aircraft, not all aircraft within a circle around a location.

If I misunderstood and it does do this, apologies, but the documentation is not clear.

wiedehopf, 2023-02-01:
```
/?closest=<lat>,<lon>,<radius in nmi>
```

SingingDwarf, 2023-02-01:
Yup, this returns multiple aircraft if they are within the radius - not just the single closest.

SingingDwarf, 2023-02-01:
Ah thanks, I was using 'circle' - apologies, I did not see the 'closest' option. Thank you.