wiedehopf/tar1090 #222: Parse aircraft image within set radius via URL

ScubyG, 2023-02-15:
Is it possible to parse the image of the nearest aircraft ie via the URL, I have looked at the readme of the URL configs and was unable to find filter by distance, only filter by altitude, another option would be to obtain the image from auto select.

What I wish to achieve is to have the nearest aircraft to my location as my desktop background which will dynamically change when another aircraft is nearer or the images cycle within a set radius.

An example in my case is URL http://10.0.0.99/tar1090/?DistanceMax=25

Thank you, the software works great and have been using it in place of Ff24 but decided to add them as a feed in addition to adsexchange.

RundesBalli, 2023-02-15:
The images are fetched from https://www.planespotters.net/

You could get the HEX/Reg from tar1090 and use the planespotters api:
https://www.planespotters.net/photo/api

Just my thoughts to this.

ScubyG, 2023-02-15:
> The images are fetched from https://www.planespotters.net/
> 
> You could get the HEX/Reg from tar1090 and use the planespotters api:
> https://www.planespotters.net/photo/api
> 
> Just my thoughts to this.

This is what I was thinking of doing, its just how to  obtain the HEX or Reg from tar1090 referenced to the closest aircraft or aircrafts within a set radius.

wiedehopf, 2023-02-16:
You can actually enable the readsb api port and do http queries.

closed 2023-02-16 by wiedehopf