wiedehopf/tar1090 #225: Add ability to see most recent flight on map

rickstaa, 2023-02-17:
I needed to find a way to get the most recent leg of a plane by cycling through the history. It would be nice if the interface accepted a `history=latest` query that shows the latest known leg for each plane in the `?icao=` query. Let me know what you think. I currently do not have time to look at this, but I wanted to put this here so that I don't forget.

wiedehopf, 2023-02-17:
I have not plans on working on this currently.
It will require changes to have a directory per aircraft of which days there are data for.

rickstaa, 2023-02-17:
> I have not plans on working on this currently. It will require changes to have a dictionary per aircraft of which days there are data.

Thanks for letting me know. Can we keep this open in case I or somebody else wants to create a PR?

ggoodloff, 2023-02-21:
Have you tried the timelapse1090 install?

rickstaa, 2023-02-21:
> Have you tried the timelapse1090 install?

Good tip, thanks!

wiedehopf, 2023-02-22:
No that's a shit tip.

rickstaa, 2023-02-22:
> timelapse1090

Ah, thanks for the update. I still need to check it out thoroughly. The goal of that repository differs from what I'm trying to do, but I thought there might be code I could use to create a PR in this repo.

ggoodloff, 2023-02-22:
It was a thought.