wiedehopf/tar1090 #226: Import static aircraft json

ggoodloff, 2023-02-21:
It's there a way to import a static aircraft json file from an API and have it show a screenshot screen (frozen) of that information to use as a bot screen capture.  I want to be able to utilize a properly formatted flight API JSON file to put a spotter bot in another area

Or is there a way to import this file into my readsb to see the flights in the map to fill in gaps I have in coverage where I am?

Located at https://api.adsb.lol/v2/all

wiedehopf, 2023-02-22:
Just don't run readsb.
Just make the /run/readsab dir and place a receiver.json that is sanely formatted.
Or make an tar1090 instance that uses a different /run dir.

Then place the aircraft.json

ggoodloff, 2023-02-22:
Thanks. I will definitely give that a shot

ggoodloff, 2023-02-22:
So I could never get the API data to load correctly, which is fine now
because they are pushing their beast port to me.

Here's my problem now, using readsb - my tar1090 display is only a 450mile
max.

Here's the kicker, on a separate system I'm using combine on, pulling from
this 450mi machine, I get an entire globe of data.

I've been at this for about 6 hours going through ever file I can find that
mentioned a range. I've reinstalled tar1090 and still just a 450mi range.

ggoodloff, 2023-02-22:
Figured out out.  Need to set --max-range 99999

I just erased the option thinking it would eliminate it.

Thanks for your amazing programs!

closed 2023-02-23 by wiedehopf