wiedehopf/tar1090 #227: Missing argument in Url query?

ggoodloff, 2023-02-28:
Am I missing an argument in my URL query that prevents the colorful altitude graph on the bottom from being displayed? I'm using a screen shot option on a headless chrome setup.

My URL query is currently:
&iconscale=1S&hideButtons&hidesidebar&screenshot&noIsolation&zoom=11&centerReceiver

On a Rpi I get the graph, on a Linux install I don't.

Really stumped:
[image]
[image]

mpentler, 2023-03-24:
That is usually set by the settings cogwheel and I'm not sure if there's a url parameter for it.

wiedehopf, 2024-05-23:
```
&altitudeChart=0
&altitudeChart=1
```

closed 2024-05-23 by wiedehopf