wiedehopf/tar1090 #228: Range rings madness'

Puppa-Smurf, 2023-03-09:
Ok, here we go.

I have Tar1090 and Readsb running, and I have edited the config.js file shown below:

./usr/local/share/tar1090/html/config.js

> // Also called range rings :)
> SiteCircles = true; // true to show circles (only shown if the center marker is shown)
> // In miles, nautical miles, or km (depending settings value 'DisplayUnits')
> SiteCirclesDistances = new Array(10,15,20);
> // When more circles defined than cirle colors last color will be used or black by default
> SiteCirclesColors = ['#FF0000', '#0000FF', '#00FF00'];
> // Show circles using dashed line (CAUTION, can be slow, especially when zooming in a lot)
> //SiteCirclesLineDash = [5, 5]; // null - solid line, [5, 5] - dashed line with 5 pixel lines and spaces in between

I have run inspect in chrome and confirmed that the ./usr/local/share/tar1090/html/config.js is being pulled, and all the infomation contained is ok, and the listing above is copied from the inspect output of the config.js file.

I did the same thing for ./usr/local/share/dump1090-retro-html/config.js which worked as expected, but I just cannot get Tar1090 to work ok with the rang rings. I do get a base station location on the map, I get actual range, but I just cannot configure the range rings.

Lastly I have used the lates files from this site to do the install and am now totaly at a loss

wiedehopf, 2023-03-09:
It's a layer, you might have deselected it?

wiedehopf, 2023-03-09:
Try an incognito tab.

Puppa-Smurf, 2023-03-09:
> It's a layer, you might have deselected it?

Thank you, yes I know and I can slect it and deslect it in Tar1090, but I can't change the range of the rings :(, that the issue

Puppa-Smurf, 2023-03-09:
> Try an incognito tab.

I have and no difference, Like I siad It works in you dump1090 retro, (where I can change the distance if the rings) but not in tar1090. and also I have had the web page debugger in chrome up to check that I have edited the correct config.js file, and that I can see that the range ring setting have been se correctly. :(

wiedehopf, 2023-03-09:
I've tried that exact config block it works fine for me.

Puppa-Smurf, 2023-03-09:
Can see any major errors

I was going to pose a picture of the console, but you can do it here :(

Puppa-Smurf, 2023-03-09:
> Can't see any major errors
> 
> I was going to pose a picture of the console, but you can do it here :(

I have found a file "defaults_9ed59fb.js" that is getting passed and my guess is that this is being used insted of the config.js file that I edited. I'll try and find it on the pi!

wiedehopf, 2023-03-09:
It's the defaults, the config.js is applied after to overwrite the defaults.

Delete config.js, run the install script, see if the range rings work.

wiedehopf, 2023-03-09:
Then modify line by line the relevant stuff.

Puppa-Smurf, 2023-03-09:
> It's the defaults, the config.js is applied after to overwrite the defaults.
> 
> Delete config.js, run the install script, see if the range rings work.

Yep, its taking the "defaults_9ed59fb.js" that being used, and for some resion the config.js is not being used. 

I have just edited the "defaults_9ed59fb.js", its taking the "defaults_9ed59fb.js" and it worked first time ;) 

I have checked the config.js and its fine and not corrupted. 

but I have it working by editing the "defaults_9ed59fb.js" so...... now just have to work out why the config.js in not doing its job :(

wiedehopf, 2023-03-09:
Do what i wrote.
Reset to known state.
You screwed something up obviously or it would load config.js.

closed 2023-03-09 by wiedehopf