wiedehopf/tar1090 #246: error after apt upgrade

loewal, 2023-05-09:
Hi,
I did an apt update and apt upgrade.
Following was upgraded:

The following packages will be upgraded:
  libcurl3-gnutls libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 linux-libc-dev raspberrypi-bootloader raspberrypi-kernel

After a reboot readsb is not starting anymore.
The error in the journal is:

May 09 15:04:43 adsb-pihole readsb[2370]: invoked by: /usr/bin/readsb --gain -10 --lat 50.887152 --lon 5.995650 --device-type rtlsdr --ppm 0 --max-range 450 --db-file=none --net --net-heartbeat 60 --net-ro-size 1200 --net-ro-interval 0
May 09 15:04:43 adsb-pihole readsb[2370]: [2023-05-09 15:04:43.440 CEST] readsb starting up.
May 09 15:04:43 adsb-pihole readsb[2370]: readsb version: 3.14.1604 wiedehopf git: c4a0e40 (committed: Thu May 4 21:41:31 2023 0200)
May 09 15:04:43 adsb-pihole readsb[2370]: Using lat:   50.8872, lon:    5.9957
May 09 15:04:43 adsb-pihole systemd[1]: readsb.service: Main process exited, code=killed, status=7/BUS
May 09 15:04:43 adsb-pihole systemd[1]: readsb.service: Failed with result 'signal'.
May 09 15:04:44 adsb-pihole vncserver-x11[595]: AgentInitCheck: agent comms failure
May 09 15:04:44 adsb-pihole adsbexchange-mlat[588]: Beast-format results connection with ::1:30104: [Errno 97] Address family not supported by protocol
May 09 15:04:46 adsb-pihole systemd[1]: systemd-hostnamed.service: Succeeded.
May 09 15:04:49 adsb-pihole vncserver-x11[595]: AgentInitCheck: no response from agent
May 09 15:04:49 adsb-pihole sshd[2445]: Accepted password for root from 10.0.0.2 port 52992 ssh2
May 09 15:04:49 adsb-pihole sshd[2445]: pam_unix(sshd:session): session opened for user root by (uid=0)
May 09 15:04:49 adsb-pihole systemd[1]: Created slice User Slice of UID 0.
May 09 15:04:49 adsb-pihole systemd[1]: Starting User Runtime Directory /run/user/0...
May 09 15:04:49 adsb-pihole systemd-logind[376]: New session 7 of user root.
May 09 15:04:49 adsb-pihole systemd[1]: Started User Runtime Directory /run/user/0.
May 09 15:04:49 adsb-pihole systemd[1]: Starting User Manager for UID 0...
May 09 15:04:49 adsb-pihole systemd[2594]: pam_unix(systemd-user:session): session opened for user root by (uid=0)
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on GnuPG network certificate management daemon.
May 09 15:04:49 adsb-pihole systemd[2594]: Reached target Paths.
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on GnuPG cryptographic agent and passphrase cache.
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
May 09 15:04:49 adsb-pihole systemd[2594]: Condition check resulted in Sound System being skipped.
May 09 15:04:49 adsb-pihole systemd[2594]: Reached target Timers.
May 09 15:04:49 adsb-pihole systemd[2594]: Starting D-Bus User Message Bus Socket.
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
May 09 15:04:49 adsb-pihole systemd[2594]: Listening on D-Bus User Message Bus Socket.
May 09 15:04:49 adsb-pihole systemd[2594]: Reached target Sockets.
May 09 15:04:49 adsb-pihole systemd[2594]: Reached target Basic System.
May 09 15:04:49 adsb-pihole systemd[2594]: Reached target Default.
May 09 15:04:49 adsb-pihole systemd[2594]: Startup finished in 127ms.
May 09 15:04:49 adsb-pihole systemd[1]: Started User Manager for UID 0.
May 09 15:04:49 adsb-pihole systemd[1]: Started Session 7 of user root.
May 09 15:04:51 adsb-pihole tar1090[593]: No aircraft.json found in /run/readsb during the last 30 seconds! Try restarting dump1090 or reinstalling tar1090 if you switched dump1090 to readsb!
May 09 15:04:51 adsb-pihole tar1090-ax978[616]: No aircraft.json found in /run/adsbexchange-978 during the last 30 seconds! Try restarting dump1090 or reinstalling tar1090 if you switched dump1090 to readsb!
May 09 15:04:52 adsb-pihole adsbexchange-mlat[588]: Connection to 127.0.0.1:30005 lost: [Errno 111] Connection refused
May 09 15:04:52 adsb-pihole adsbexchange-mlat[588]: Reconnecting in 10.6 seconds
May 09 15:04:54 adsb-pihole vncserver-x11[595]: AgentInitCheck: agent comms failure

Can you see what has happened? A bug?

Thank you!

wiedehopf, 2023-05-09:
It's a bug with debian libzstd1 package.

Don't really know how to fix it.

You want to try some stuff? If yes, read on, otherwise disregard the rest of the message.

First reinstall readsb: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

Show the output for the following commands:
uname -m
dpkg --print-architecture

loewal, 2023-05-09:
Hi,

This is the output:

***@***.***:~/Downloads # uname -m
armv7l
***@***.***:~/Downloads # dpkg --print-architecture
armhf
***@***.***:~/Downloads #

These commands I executed in a working environment (image)and without those upgrades.

Regards

Loe

wiedehopf, 2023-05-09:
Just for correctness, the issue is with: https://github.com/wiedehopf/readsb

loewal, 2023-05-09:
Yes, I reran the script you mentioned earlier.
I did that several times before, but when I did the apt upgrade and rebooted there was that error that connection wasn’t possible anymore.

Which of them should NOT be deployed? (libcurl3-gnutls libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 linux-libc-dev raspberrypi-bootloader raspberrypi-kernel)

Regards

RundesBalli, 2023-05-09:
> Just making an entry to keep myself abreast of this issue.

[image]

wiedehopf, 2024-05-11:
Identical to this, see workarounds there: https://github.com/wiedehopf/readsb/issues/67

closed 2024-05-11 by wiedehopf