wiedehopf/tar1090 #254: Interfacing tar1090 with dumphfdl via readsb - success but very few icons on map

ct2iww, 2023-06-19:
Hello!

I am trying to get tar1090 to receive aircraft positions from dumphfdl, via readsb, which I believe I have managed to do, with some sucess. 

However, very few positions are displayed on the map, although the aircraft show on the list. The system runs on a 64 bit PC, under a version of Debian called DietPi.

I had to edit the install script for tar1090 to read aircraft.json from a specific folder, in my user home. I had to do this (and instruct readsb to write there, as well) since the install script was aborting with an error message, allerting that it was not finding the usual folders, namelly the one for reasb, although that program was installed and running.

Second, or let´s say paralell, step, was to instruct, via the startup comand, for readsb to write aircraft.json to a specific folder and open a connection to the port where my instance of dumphfdl will send the basestation output:

readsb --net-only --net-sbs-in-port=20005 --write-json /home/dietpi/json

Third step is to start dumphfdl to send basestation output to a specific local port:

dumphfdl --soapysdr driver=sdrplay --sample-rate 250000 8834 8885 8894 8912 8927 8939 8942 8948 8957 8977 --output decoded:basestation:tcp:address=localhost,port=20005.

I was able to get everything running, both dumphfl and readsb aknowledge a connection, the aircraft list is populated, but very few aircraft are showing on the map. I am also under the impression that a lot of RX aircraft dont even make it into tar1090.

[image]

Upon further reading, I came across a post that suggested that the receiver´s  location has to be set up either on readsb or on tar1090, to solve that problem, with my understanding that readsb´s information supercedes tar1090´s:

sudo readsb-set-location xx.xxxx -x.xxxx

I also edited tar1090´s config.js with the same information, and also come customization:

 sudo nano /usr/local/share/tar1090/html/config.js

So, the changes do reflect on the map, like the receiver location and custom name, the received aircraft do show on the list, but only occasionally does an aircraft icon appears on the map.

So, what am I doing wrong? I suspected that there would be a range limit somewhere to be determind, but I do get some flights to show up many thousands of km from my location.

Also, is reading the aircraft.js list the only way to exchange info between the programs? I tried with a re-output in basestaion format and getting readsb to read from there, but either got it wrong or just doesnt work that way.

I´d appreciate your thoughts on that.

Paulo

wiedehopf, 2023-06-19:
--max-range 0

ct2iww, 2023-06-19:
> 

--max-range 0, in readsb, I guess? Tried it, but didnt make a change. Readsb is, as far as I can understand, getting the positions correctely from dumphfdl:
*;
hex:  899000   CRC: 000000 fixed bits: 0 decode: ok
reduce_forward: 1
receiverTime:                        0.00us
utcTime: 15:20:41.405 epoch: 1687188041.405
SBS  addr: 899000
  ICAO Address:  899000 (other)
  Air/Ground:    airborne
  Ident:         CAL081
  Latitude:      51.667083
  Longitude:     2.016682

*;
hex:  899000   CRC: 000000 fixed bits: 0 decode: ok
reduce_forward: 0
receiverTime:                        0.00us
utcTime: 15:20:41.406 epoch: 1687188041.406
SBS  addr: 899000
  ICAO Address:  899000 (other)
  Air/Ground:    airborne
  Ident:         CAL081
  Latitude:      51.667083
  Longitude:     2.016682

*;
hex:  01012a   CRC: 000000 fixed bits: 0 decode: ok
reduce_forward: 1
receiverTime:                        0.00us
utcTime: 15:20:51.218 epoch: 1687188051.218
SBS  addr: 01012a
  ICAO Address:  01012A (other)
  Air/Ground:    airborne
  Ident:         MSR778
  Latitude:      51.059744
  Longitude:     3.309638

*;
hex:  01012a   CRC: 000000 fixed bits: 0 decode: ok
reduce_forward: 0
receiverTime:                        0.00us
utcTime: 15:20:51.219 epoch: 1687188051.219
SBS  addr: 01012a
  ICAO Address:  01012A (other)
  Air/Ground:    airborne
  Ident:         MSR778
  Latitude:      51.059744
  Longitude:     3.309638

result in tar1090:

[image]

wiedehopf, 2023-06-19:
--net-sbs-jaero-in-port

wiedehopf, 2023-06-19:
That circumvents all the stuff you did starting everything manually ... 

I couldn't really be bothered reading everything sorry, i'll just keep throwing my best guess at it :)

ct2iww, 2023-06-19:
I think that did it, thanks for taking the time, sorry for the long post.
Will monitor the output for the next few hours just to be sure.

Paulo

wiedehopf, 2023-06-19:
Was good to give the detail i skimmed it :)

Good day.

ct2iww, 2023-06-21:
Thank you for the promt answers, I guess that I have it working rather well. The only one thing that I need to adjust is the time on screen and list, it seems to be set at 2.000 seconds, or close to 35 minutes, which is too long. Looks like on HFDL airplanes just dont TX as often as ADSB. I´ve looked into tar1090´s config.js, saw a couple lines in the end pertaining to timeouts, but it doesn´t seem to the place to ajust that. Any thoughts? 
Thank you.

wiedehopf, 2023-06-21:
https://github.com/wiedehopf/readsb/blob/dev/help.h

It's not set at 2 seconds, not sure how you get that. or you mean 2000?
2.000 is 2 decimal 000, careful with the 1000 separator in international use ;)

readsb --help will show you how to change it.

mpentler, 2023-12-04:
I've just tried this myself and my normal ADS-B traffic is timing out in 60 seconds while the HFDL traffic is tagged as Sat ADS-C and is timing out in around 35 minutes...

closed 2024-09-07 by wiedehopf