wiedehopf/tar1090 #258: Issue with 'seenTimeout'?

gvanem, 2023-07-25:
I use tar1090 as the web-page in my own [Dump1090](https://github.com/gvanem/Dump1090) program. Take a look at this Win-10 screen-shot:
[image]

I have a timeout of 60 sec before I remote a plane in interactive-mode. And thus I **do not** include this in the `aircrafts.json` response. But in tar1090, a plane gets removed approx. 30 sec after I remove it from the console; 4 planes in tar1090 and 3 in mine.

I see in `config.js` a `seenTimeout = 58; // in seconds` which should be the default value, no?
So I assume there is a timer that restarts when a plane-object gets received in `aircrafts.json`. This is an example
of what I send  for 2 planes (edited for clarity):
```json
{"now":1690268461.333, "messages":8033, "aircraft":
 [ {"hex": "471f87", "flight": "WZZ7176", "lat": 60.3153, "lon": 5.44098, "altitude": 8525, "track": 122,
    "speed": 292, "type": "adsb_icao", "messages": 692, "seen": 2, "seen_pos": 1},
   {"hex": "47c1ab", "flight": "", "lat": 60.2278, "lon": 5.06126, "altitude": 2525, "track": 104,
     "speed": 133, "type": "adsb_icao", "messages": 37, "seen": 2, "seen_pos": 1}
]}
```

Does it look OK or is there something missing? What's the relationship between the sent `"now"` value and this `seenTimeout`?
Maybe this time-lag is due to my misunderstanding of my Dump1090's response.
Please advice.

wiedehopf, 2023-07-25:
I don't know the default and i don't care for use with non standard programs quite frankly.
In other words, feel free to debug the js however you see fit.

seenTimeout has some interactions with other code .... not sure what the effective timeout is.
planeObject.js ... should have the effective seenTimeout.

gvanem, 2023-07-25:
I seems to have misunderstood the logic behind this timer. A fix for me is simply to *retire* the plane quicker for a weeb-client
than in interactive mode.

closed 2023-07-25 by gvanem

wiedehopf, 2023-07-25:
It's just bad logic sorry ;)
The name implies a fixed timeout, but i applied modifiers as the code progressed and different modes were added.