wiedehopf/tar1090 #263: Vietnam Flag's color is not the correct Red

magihub, 2023-09-07:
I found that Vietnam Flag's color is not the correct Red, Could you please replace a better one? Thanks.

rcarsey1, 2023-09-07:
I concur.  Attached is a new PNG to use for replacement.
[image]

magihub, 2023-09-07:
Thanks Very Much, I replaced this local png file in /usr/local/share/tar1090/html/flags-tiny folder, and it‘s OK after I clear the browser cache.  Maybe you should replace this one ( https://github.com/wiedehopf/tar1090/tree/master/html/flags-tiny/Vietnam.png ) at the same time.

ngtrthanh, 2023-10-20:
@magihub, @rcarsey1 I have a complete new set of flags here
https://github.com/ngtrthanh/docker-readsb-protobuf/pkgs/container/docker-readsb-protobuf
https://github.com/ngtrthanh/readsb-protobuf/tree/dev/webapp/src/images/flags-tiny
[image]