wiedehopf/tar1090 #283: H[ome] on Safari/iOS can require two presses to get to home location

nbriggs, 2023-12-21:
When you touch "H", and iOS prompts for permission to provide location information, the map display is moved to the NY area *before* permission is given and does not move to the home location *after* permission is given, necessitating a second press of the "H" button to redisplay at the desired location.

mpentler, 2024-05-26:
Have also seen this, have you seen it every single time? I'm not sure I have had it happen on every occasion.

nbriggs, 2024-05-26:
@mpentler it may indeed not be every single time, but it's certainly most of the time for me.  It'll typically be first thing in the AM after the phone has been idle overnight and I switch to the tab with the virtual radar display running, but the home location dot has disappeared.

wiedehopf, 2024-05-26:
What site exactly do you have open?

Also you should really be able to allow the domain to permanently access the location.

nbriggs, 2024-05-26:
https://globe.adsbexchange.com/

I probably could set it to Allow, but I have it set to Ask.

wiedehopf, 2024-05-26:
First of all that code is not the same as this code anymore.
For some time.
So even if i were to make changes, would take quite some time to possibly get merged over there.

And yes you could just allow it.
That would mean it would just keep updating the even if you move.
Pretty much that's the two use cases i considered: either you deny it forever and just not use the H button, or you allow permanently and it keeps updating the device location.

Maybe i test your use case at some point on android, i guess i should be able to reproduce it.

nbriggs, 2024-05-26:
I don't know how Android behaves with regard to the location when location permission is denied.  In fact, I'm not sure how iOS behaves either: whether it gives a fake location, or returns an error (or throws an exception) when asked.  On the application side - which you know and I don't - it could be using a default location (and influencing the framing) if denied rather than the current framing of the map just without the current location dot.

wiedehopf, 2024-05-27:
So while this is fixed in my branch, i have no control over adsbexchanges website.

If you supply data, consider some variety in sites to feed, for example using this image: https://adsb.im/

I'd expect this change to propagate soon to some of the other websites that use this interface (for example https://adsb.lol/)

closed 2024-05-27 by wiedehopf

nbriggs, 2024-05-27:
Nice, thanks!  I'll see if I can prod the ADSBExchange people to update.  I'm not a feeder - I live in the SF Bay area where there seems to be very good coverage already.