wiedehopf/tar1090 #287: [Feature] Offline Images

SenorFusion, 2024-02-13:
First, just want to say thank you for the phenomenal software! I did a complete install from a brand new Pi Zero 2 to fully functional ADS-B station in less than an hour. The install scripts are amazing and every single one worked flawlessly including graphs, auto-gain, timelapse, everything!

Is it possible to download something like the planespotters.net database and have it work on tar1090 offline?

My station will be running completely offline and I was hoping to still be able to get pictures.

I realize that could be an insanely large database and might be out of data kind of quickly.

I've tried to search if this has been done, but Google is failing me. I didn't find anything in the history of your repos here either.

Thanks!

RundesBalli, 2024-02-13:
Even though I am not the maintainer of this project, I can still contribute my small part to your issue, especially concerning the Planespotters.net topic: Saving the photos/data of Planespotters is forbidden according to ToS. Content must be integrated directly from Planespotters.net, which is why I assume that there is no "ready-made package" available with all the photos/data for download. In addition, with the amount of photos, this would certainly fill several hard drives.

https://www.planespotters.net/photo/api

```
Photo API Terms of Use
(...)
Image files must not be stored on your servers or inside your application and must be served from the original URL provided in the API response. 
```

wiedehopf, 2024-02-13:
I can leave this open like many other issues but i mostly maintain and do not add functionality to this project.

closed 2024-05-11 by wiedehopf