wiedehopf/tar1090 #288: [Feature] 3D map / Streets-GL

JoranJix, 2024-02-14:
I wonder if it would be possible to add an optional view with streets-gl https://github.com/StrandedKitty/streets-gl.
I think for some people it could be interesting to see a plane or plane in 3D space around their region. 
It would make a great addition and bring the project a big leap forward. 
[image]

SammySkycrafts, 2024-03-03:
Having a dedicated service providing 3D viewing would likely result in better performance. Something like https://store.steampowered.com/app/2789510/FlightTracker3D/ perhaps.

gvanem, 2024-03-17:
Chrome uses 2 GByte of RAM to show my airport.
[image]

Yikes!