wiedehopf/tar1090 #298: Custom Tile / Map Server

adamstern2000, 2024-04-07:
I carry a tile server for field work.   I tried to directly modify the html/layers file, but it doesn't seem to make any changes to the web interface.  Is there any way to add a custom tile server - it's url is very similar to an OSM server.

wiedehopf, 2024-04-08:
clone the git, then:
```
./install.sh test
```
That will install with local modifications

wiedehopf, 2024-04-08:
```js
227 const customTiles = usp.get('customTiles');
228 if (customTiles)
229     loStore['customTiles'] = customTiles;
230 if (customTiles == 'remove')
231     loStore.removeItem('customTiles');

 27     if (loStore['customTiles'] != undefined) {
 28         custom_layers.push(new ol.layer.Tile({
 29             source: new ol.source.OSM({
 30                 "url" : loStore['customTiles'],
 31                 maxZoom: 15,
 32                 transition: tileTransition,
 33             }),
 34             name: 'custom_tiles',
 35             title: 'Custom tiles',
 36             type: 'base',
 37         }));
 38     }
```

This is in the code ... so you can do stuff with that within the limitations of that code.

Append ?customTiles=URL to the tar1090 URL and see if that works.

adamstern2000, 2024-04-08:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Thanks buddy!

Thank you,

Adam Stern
President
d:760.585.4084
t: 760.585.4090
e: ***@***.******@***.***>
a: Infinitely Virtual
    1920 Hillhurst Avenue
    Suite 290
    Los Angeles, CA  90027
w: www.infinitelyvirtual.com<http://www.infinitelyvirtual.com/>

wiedehopf, 2024-04-11:
Why didn't you try the URL i suggested?
It injects it locally into the page settings for the browser.

```Append ?customTiles=URL to the tar1090 URL and see if that works.```

wiedehopf, 2024-04-11:
Well check the dev console what's wrong.

Possibly cross origin stuff.
Really this is for people to debug it themselves.
I might continue giving hints though :)

wiedehopf, 2024-04-11:
URLs start with http

wiedehopf, 2024-04-11:
/tar1090/?customTiles=http://10.0.0.2:83/default/{z}/{x}/{y}.png

RundesBalli, 2024-04-12:
> I managed to fix it, it was a cross origin access error. Thanks for the help!

Normally you would share your solution here now, so that other people with the same problem don't have to search for the solution in the future, but can adopt your solution.

Nekzuris, 2025-07-11:
There is an issue with URL that have `&` even when encoded with `%26` like:
http://adsb-feeder.local:8080/?customTiles=https://mt0.google.com/vt/lyrs=s,h%26x={x}%26y={y}%26z={z}

`&` is not present in the tile request.

wiedehopf, 2025-07-11:
Someone raised issues with XSS, so some chars are removed from URL parameters.
& is among them.

I don't recall the actual issue so i'm a bit hesitant to just remove it.
Also this is technically a different issue ....