wiedehopf/tar1090 #300: Planes are Mode S when beyond ~500 miles

alexkuck, 2024-04-13:
I travel with my ADSB receiver. When I go beyond ~500 miles from my `readsb-set-location` all planes are listed as "Mode S" only and I see only altitude, track, and speed. When I return to within ~500 miles of my `readsb-set-location`, the planes reappear as ADS-B and I see position data again.

Is this intended functionality?

wiedehopf, 2024-04-13:
--max-range 0

closed 2024-04-13 by alexkuck

wiedehopf, 2024-04-13:
readsb also has a gpsd input if you care for an automatically updating tar1090 map.

This needs to be added to the readsb configuration:
```
--net-connector localhost,2947,gpsd_in
```

You can press H on the tar1090 map and it will follow the changing location automatically.
The dot and rings will be updated independently of pressing H.