wiedehopf/tar1090 #302: How to get registration/callsign via json webinterface

nils10, 2024-04-22:
Hi, I am running readsb with tar1090 locally and am using the aircraft.json to query data with an external application.

I noticed the aircraft.json doesn't seem to include the registration or callsign of the aircraft, even though on the tar1090 webinterface its shown so I assume its available in the backend. How can I get the registration/callsign via the json file?

Thanks in advance for any feedback.

nils10, 2024-04-22:
Answering my own question: Got it to work by adding in the /etc/default/readsb config file: --db-file /usr/local/share/tar1090/aircraft.csv.gz" to the RECEIVER_OPTIONS line. The complete line in my case then looked like this:
`RECEIVER_OPTIONS="--net-only --write-json-every 1 --db-file /usr/local/share/tar1090/aircraft.csv.gz"`

I've also tried adding --db-file-lt /usr/local/share/tar1090/aircraft.csv.gz to the same line, but that resulted in the following error, so I left that out for now:
tar1090[796]: No aircraft.json found in /run/readsb during the last 30 seconds! Try restarting dump1090 or reinstalling tar1090 if you switched dump1090 to readsb!

wiedehopf, 2024-04-22:
Read the log, read the help for readsb: readsb --help

closed 2024-04-22 by nils10