wiedehopf/tar1090 #304: how do i debug my client

Tokyo-All-Ready, 2024-04-25:
i installed: https://github.com/antirez/dump1090
ran: ./dump1090    its working
opened up tar1090 at:  http://192.168.1.23/tar1090/ / http://127.0.0.1/skyaware978/ is installed and working but no messages being decoded.

is there a debug log i can work with?

i had to blacklist echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf for it to work
and i assume a reboot is required.

but i need a debug log so i know the error i have to fix

wiedehopf, 2024-04-25:
https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

That's a very old dump1090, use the guild linked above.
1090 + 978 will need 2 separate SDRs.

Tokyo-All-Ready, 2024-04-25:
yes thank you that totally fixed it  :)

https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb
i got adsb working and 978 decoder working. 

i got offline maps but is there a way i can download the entire earths openstreetmaps (its like 80 or 90 gb) and just have tar1090 use that?  and how do i use this program without any internet.

wiedehopf, 2024-04-25:
No.

closed 2024-05-11 by wiedehopf