wiedehopf/tar1090 #306: [Feature]

jwalatham, 2024-04-26:
It would be great to utilise the Openlayers ruler code, to draw a line between two points on the map and measure the distance and bearing between them. For example, to measure how far an aircraft is from my location or another point, or the distance between two aircraft.

RundesBalli, 2024-04-26:
Very nice idea, I support this feature request. :-)

mpentler, 2024-05-26:
I've wanted to try and do this for a while but couldn't figure it out. It'd be like ATC staff have on their displays and would be cool for better situational awareness. I'd love to see this but foresee it being difficult to integrate with scrolling and map clicking. Maybe a button up top to enter that mode?

mpentler, 2024-05-27:
Well, I started to learn how to make a layer and draw on it, and now I've added a ruler button that turns the layer on and off and draws a dot at 0,0.

Baby steps. This is helping me to learn the tar1090 flow too. I need to figure out where to add some click handlers to be able to draw on the map, though.

wiedehopf, 2024-05-28:
Really one issue is where to put an extra button.

I suppose the extra button can be enabled from the settings menu ...

mpentler, 2024-05-28:
You're right. I am not running globe mode on my local install, so I had no replay button and could stick it there. Otherwise it'd just have to go next to it, and then that's 5 buttons on the top row U/H/T/Replay/Ruler

Other option is a button on the top left, somewhere untouched by other things right now. Although my ruler controls are currently displayed up there when you turn the ruler mode on. Could make it a drop down thing - there's too many possibilities with this. I am trying to match the design of the map as-is right now.

mpentler, 2024-06-03:
Got a working prototype going, but I don't think my code is really up to scratch. I will probably put in a branch on my own fork.

[image]
[image]

mpentler, 2024-06-08:
Now up on my repo, ruler branch. It's very shit.

Telecomeansa, 2025-05-15:
Good morning, how are you? I'm interested in implementing distances between aircraft, please help me.