wiedehopf/tar1090 #307: ADSB daily report showing low altitude near me

Mike-D-424, 2024-04-27:
Is there a way for me to generate a daily report of aircraft near me (I have a pi feeder) below 1000 ft?

I am tired of low-flying helicopters shaking my house and wish to get a report showing the aircraft number, time, date, altitude, heading, etc. 

My skill level is limited therefore a very easy-to-implement solution is preferred.  

Thank you!

Michael

RundesBalli, 2024-04-27:
Not exactly what you want, but you can send yourself a message when an aircraft passes over a certain area:
https://github.com/RundesBalli/adsbTelegramNotifier

Maybe this helps

Mike-D-424, 2024-05-10:
Thank you,  I am not smart enough to get this installed, configured, or 
to get reports generated.

near me (Issue #307)

closed 2024-05-10 by wiedehopf