wiedehopf/tar1090 #315: web interface not available 404 /tar1090

nessnah198, 2024-06-04:
Hello, i've been struggling with accessing the web interface, only get 404. 
Funny thing is i have another machine running debian, did the same setup and it worked perfectly.

How do I debug this?

I have installed readsb/tar1090 with:
* sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"
  * no web interface
* then: sudo bash -c "$(wget -nv -O - https://github.com/wiedehopf/tar1090/raw/master/install.sh)"
  * no web interface

Tried to reinstall but no difference.
If i run flightmon from my PC accessing my server with --dump_host option I get data.
lighttpd is installed.

wiedehopf, 2024-06-04:
What URL are you trying?
Try curl http://localhost/tar1090 and see what you get
Check the lighttpd log?

nessnah198, 2024-06-04:
```
server$ curl http://localhost/tar1090
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL was not found on this server.</p>
<hr>
<address>Apache/2.4.59 (Debian) Server at localhost Port 80</address>
</body></html>
```

Curl only gives me 404 and /var/log/lighttpd/ is empty.. hmm

wiedehopf, 2024-06-04:
lighttpd logs to journalctl

also that's apache and not lighttpd in the 404
put lighttpd on a different report or remove apache.

nessnah198, 2024-06-04:
You're right!
I stopped apache2 and ran the bash script again.
Apache was not installed on my second computer..

Thanks for your help🛩️

closed 2024-06-04 by wiedehopf