wiedehopf/tar1090 #318: Adding new/additional map sources?

mzdial, 2024-06-27:
Hi!

Brand new user to tar1090. Been trying to figure out how to add additional new maps/tiles to the application? This was spurred on by running across this implementation, using "Stadia_Alidade_Smooth" as the map/tiles.

https://skylink.hpradar.com/

I'm wanting to add this "Stadia_Alidade_Smooth" style to my personal tar1090 instance. How do you go about doing this?

Thanks!
Matt

bfayers, 2024-08-17:
You would modify your `layers.js`

You can see how that site currently has their layers configured by looking at their layers.js file here: https://skylink.hpradar.com/layers_4ef9616bfcad8ef3b77918d0050091d0.js  ([web archive version of link](https://web.archive.org/web/20240706233422/https://skylink.hpradar.com/layers_4ef9616bfcad8ef3b77918d0050091d0.js))

closed 2024-09-07 by wiedehopf

mzdial, 2024-09-09:
Thank you! That did it for me. Appreciate the guidance.

> You would modify your `layers.js`
>
> You can see how that site currently has their layers configured by looking at their layers.js file here: [...]