wiedehopf/tar1090 #319: MLAT flights missing?

darethehair, 2024-06-30:
The active display page for tracking flights appears to have a code/color reserved for 'MLAT' tracking, but I never see any showing up -- despite having a working MLAT-sharing configuration.  Have I forgotten to do something?

In previous alternate tar0190 setups that I have run over the years, it seems to me that my range with MLAT activated was further than I am now getting, so I am curious if I need to tweek something.

anoo2niem, 2024-07-02:
https://github.com/ADSBexchange/feedclient

darethehair, 2024-07-03:
> https://github.com/ADSBexchange/feedclient
> 
> Edit: sorry, i read over it to fast and just see u already had a working MLAT sharing config. However leaving it here as the part 'Disable / Enable adsbexchange MLAT-results in your main decoder interface' could be the tweek u need, or set u on the right path.

Yes, MLAT is configured and the ADS-B Feeder Status page tells me that MLAT is 'connected' -- but I never see any MLAT flights in the tracking display.  I am not sure what 'enable/disable' you are referring to, but I did try to run the 'axupdate.sh' script in hopes that it might be there -- and it eventually failed with compile errors (!) on 'readsb'.  As for MLAT, it told me the following:

I```
nstalling mlat-client to virtual environment

34
36
38
40
46
48
50
60
70
```

wiedehopf, 2024-07-03:
https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

You gotta check the MLAT sync.
For example the mlat-client log will tell you more.

https://github.com/wiedehopf/adsb-wiki/wiki/mlat-check

wiedehopf, 2024-07-03:
Most aircraft have ADS-B transponders.

That sync pattern suggests limited range as well, that doesn't help either.

closed 2024-08-16 by wiedehopf