wiedehopf/tar1090 #320: Live tar interface is not working

sbparis, 2024-07-05:
I am using the tar1090 instance provided on an AWS EC2 instance. I'm new to this world, so don't rule out something silly. I'll try and provide all of the information that I feel is relevant, but if there is more information required I'm happy to provide it.

The server is acting as a sort of aggregator from multiple receivers using the feeder scripts that were modified from the adsb-feeder-scripts repository and is running on Ubuntu 24.04. The tar1090 service is clearly active and functioning in some capacity seeing as the tar1090/data/aircraft.json as well as tar1090/data/aircraft.binCraft.zst links are populated and acting as expected. However, the live interface of tar1090, or the replay functionality is not working correctly. 

[image]

To me, this indicates that tar is actively receiving the data but having trouble displaying it. When I go into chrome devtools this is what I see:

[image]

I have another physical aggregator that is working that I was using for testing before opening up the EC2 instance and it uses all of the same configuration. However, the same jquery is not returning a 404: 

[image]

I've tried checking the nginx-tar1090.conf file, as well as my tar1090 and readsb configurations but can't seem to find any issues. Could it be a security group issue on the AWS end? 

Any help would be greatly appreciated. Again, if you need more information I'm happy to provide it.

Thanks!

wiedehopf, 2024-07-05:
disable --tar1090-use-api or understand what it means :)

sbparis, 2024-07-05:
As in delete it from the readsb config?

sbparis, 2024-07-05:
Thanks so much! That did the trick.

closed 2024-07-05 by sbparis

wiedehopf, 2024-07-05:
```
readsb --help | grep api
```