wiedehopf/tar1090 #321: only showing UAT

aurochsen, 2024-07-08:
Clean install with both 1090 and 978 on the same PI. Tar1090 (and graph1090) is only showing 978 traffic. Where skyaware978 and 1090 are both showing planes. I did make the changes to the config file and ran "echo /run/skyaware978 tar1090 | sudo tee /etc/default/tar1090_instances" nothing seems to work whether the config UAT is set to yes or no I only get UAT. 
8080/airplane.json shows 33 planes but skyaware-978 shows 2. Tar1090 is only showing the 2 planes.

wiedehopf, 2024-07-08:
```
rm -f /etc/default/tar1090_instances
```
Then rerun the tar1090 / graphs1090 install.

Otherwise tar1090_instances will need 2 separate instances.

But if you just want combined data in tar1090, remove the tar1090_instances file and edit /etc/default/tar1090 with the 978 url.

Or consider running a premade image: https://adsb.im/

aurochsen, 2024-07-08:
This was exactly my issue. After removing the previous instances it works great. Thank you for the fast solution!

closed 2024-07-08 by aurochsen