wiedehopf/tar1090 #325: MLAT traffic mis-labelled as "Other" on tar1090 installed on mlat-server

rkarikari, 2024-08-03:
Hello,

I have tar1090 installed on an "mlat-server" and it filters out all modeS traffic. I believe that may be by design for an aggregator.

However, for MLAT traffic, the source is  displayed as "Other" instead of "MLAT" as is the case on the "mlat-clients" shown in the pics below....

mlt server ...
[image]

clients below....

mlat-client1 
[image]

mlat-client2
[image]

mlat-client3 etc....
[image]

Is there something (a setting perhaps) I have missed?

Thanks.

wiedehopf, 2024-08-03:
SBS / Basestation has no way to mark MLAT results, thus you need to feed it into a special port.
```
--net-sbs-in-port 30106
```
Use this for readsb and then check the startup log for the extra SBS port it opens.
Or have mlat-server open a listen port and have readsb connect using net connector
```
readsb --help | grep net-connect
  --net-connector=<ip,port,protocol>                             Establish connection, can be specified multiple times (e.g. 127.0.0.1,23004,beast_out) Protocols: beast_out, beast_in, raw_out, raw_in, sbs_in, sbs_in_jaero, sbs_out, sbs_out_jaero, vrs_out, json_out, gpsd_in, uat_in, uat_replay_out, planefinder_in, asterix_in, asterix_out (one failover ip/address,port can be specified: primary-address,primary-port,protocol,failover-address,failover-port) (any position in the comma separated list can also be either silent_fail or uuid=<uuid>)
  --net-connector-delay=<seconds>                                Outbound re-connection delay (default: 30)
```

Guess sbs_in_mlat is not mentioned for some reason.

wiedehopf, 2024-08-03:
mlat-server will only output MLAT results, nothing else.

rkarikari, 2024-08-03:
Thanks so much for the quick response.
I see new ports opened with  ..
`--net-sbs-in-port 30106`   included.

[image]

rkarikari, 2024-08-03:
Thank you so much.... that fixed it!

[image]

closed 2024-08-03 by wiedehopf

Telecomeansa, 2026-08-18:
Good morning, put in readsb --net-sbs-in-port 30106 but in the multilateration server in sudo nano /etc/systemd/system/mlat-server.service
--client-listen 31090 \
--filtered-basestation-listen 31003 \
--basestation-listen 31004 \
--filtered-basestation-connect 127.0.0.1:30107 , but I can't see anything in tar1090 in a multilateration process

wiedehopf, 2026-08-18:
what is the startup log. Maybe specify the mlat in port specifically. Are you sure your mlat server is producing results? Check with NC netcat

Telecomeansa, 2026-08-18:
root@dump1090-VirtualBox:/home/dump1090# nc -v 127.0.0.1 30106
Connection to 127.0.0.1 30106 port [tcp/*] succeeded!
^C
root@dump1090-VirtualBox:/home/dump1090# # Ver el proceso que usa el puerto 30106
sudo ss -tlnp | grep 30106
LISTEN 0      4096                       0.0.0.0:30106      0.0.0.0:*    users:(("readsb",pid=1198,fd=15))        
LISTEN 0      4096                          [::]:30106         [::]:*    users:(("readsb",pid=1198,fd=16))        
root@dump1090-VirtualBox:/home/dump1090# sudo journalctl -u mlat-server -f
ago 18 15:21:26 dump1090-VirtualBox mlat-server[1262]: 20260818 15:21:26.748 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)
ago 18 15:21:41 dump1090-VirtualBox mlat-server[1262]: 20260818 15:21:41.754 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)
ago 18 15:21:56 dump1090-VirtualBox mlat-server[1262]: 20260818 15:21:56.756 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)
ago 18 15:22:11 dump1090-VirtualBox mlat-server[1262]: 20260818 15:22:11.758 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)
ago 18 15:22:26 dump1090-VirtualBox mlat-server[1262]: 20260818 15:22:26.761 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)
ago 18 15:22:41 dump1090-VirtualBox mlat-server[1262]: 20260818 15:22:41.764 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 3 tracked)
ago 18 15:22:56 dump1090-VirtualBox mlat-server[1262]: 20260818 15:22:56.764 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 3 tracked)
ago 18 15:23:11 dump1090-VirtualBox mlat-server[1262]: 20260818 15:23:11.765 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 3 sync 3 tracked)
ago 18 15:23:26 dump1090-VirtualBox mlat-server[1262]: 20260818 15:23:26.766 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 3 sync 3 tracked)
ago 18 15:23:41 dump1090-VirtualBox mlat-server[1262]: 20260818 15:23:41.770 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 3 sync 3 tracked)
ago 18 15:23:56 dump1090-VirtualBox mlat-server[1262]: 20260818 15:23:56.770 coordinator  Status: (3 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 2 sync 2 tracked)

wiedehopf, 2026-08-18:
you need 4 synced clients for any MLAT results and all 4 clients need to see the same plane.
please don't post on other issues, open your own instead.

Telecomeansa, 2026-08-20:
Sorry, but I still don't see anything on tar1090.
ago 20 16:18:11 dump1090-VirtualBox mlat-server[8718]: 20260820 16:18:11.726 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 4 sync 4 tracked)
ago 20 16:18:26 dump1090-VirtualBox mlat-server[8718]: 20260820 16:18:26.728 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (1 mlat 5 sync 5 tracked)
ago 20 16:18:41 dump1090-VirtualBox mlat-server[8718]: 20260820 16:18:41.745 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 5 sync 5 tracked)
ago 20 16:18:56 dump1090-VirtualBox mlat-server[8718]: 20260820 16:18:56.748 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 5 sync 5 tracked)
ago 20 16:19:11 dump1090-VirtualBox mlat-server[8718]: 20260820 16:19:11.750 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (3 mlat 4 sync 5 tracked)
ago 20 16:19:26 dump1090-VirtualBox mlat-server[8718]: 20260820 16:19:26.750 coordinator  Status: (4 clients 0 bad sync) (0.00 outlier_percentage) (2 mlat 4 sync 5 tracked)

wiedehopf, 2026-08-20:
those 4 clients all need to have a distance between each other of more than 1km.

now please go away.
there is no bug, mlat-server is just not outputting anything.
and that's just up to you to figure out.
running mlat-server is for people with perserverance.