wiedehopf/tar1090 #326: 404 error after update to current version.

rkarikari, 2024-08-03:
Hello,
After updating to the latest version (3.14.1657) and restarting, I got the following error....

[image]

uninstall and reinstall did not fix the issue.

Any help would be greatly appreciated.

wiedehopf, 2024-08-03:
this has nothing to do with the version, don't confuse things and check what fails.

also completely unrelated to the normal operation of tar1090.
it fails cause you're changing stuff around.

wiedehopf, 2024-08-03:
re-api is a feature for those people who can fix such a 404 on their own.
deactivate if you can't.

closed 2024-08-03 by wiedehopf

wiedehopf, 2024-08-03:
Now you enabled --tar1090-use-api as a readsb option.
But it seems you failed to read the help for that option.

Good luck :)

rkarikari, 2024-08-03:
I was not in the least  blaming anyone . 
I was merely listing the sequence of events.   Sorry,  if I was not clear in communicating.

I obviously had it working before the update. I simply failed to consider that I'd have to (repeat that process)  with each update.
I had a vague memory of a comment somewhere ...that  the  configuration  would be  maintained with an update.

Truly sorry...... and thank you.

All is well now.

rkarikari, 2024-08-03:
Once more thanks. I do not take your time/work for granted.
Your assistance has been well appreciated.
Have a great day.

wiedehopf, 2024-08-03:
No worries.

tar1090 will write webserver configuration files on every update. (as they could have changed in the repository)
One option is to not include those files but rather copy them.

Or make an extra file to include.

The reason for not including this by default is that it should be a conscious user decision to expose readsb API to the web.
While i'm not aware of any issues it's a consideration.

Building a new site? :)

rkarikari, 2024-08-03:
Yes ...,and  I'm hoping I can get others to join in the fun. 
It's not popular in my part of the world.

Fingers crossed ... that changes soon.
:)

wiedehopf, 2024-08-03:
For better consistency .. consider changing configure.sh
```
--net-ro-interval=0.05 --net-ro-interval-beast-reduce=0.12
```
Replace the ro-interval with these two arguments.

This is about how long stays in the buffer for certain connections before a packet is sent.
ro-interval won't matter but let's just put default 50 ms
while the beast reduce is gonna use 120 ms

rkarikari, 2024-08-03:
Done!
Thank you so much.