wiedehopf/tar1090 #328: Device or resource busy:blacklist didn't work

kevin4396, 2024-08-09:
I use readsb+tar1090 to track ADSB signals with rtlsdr. 
When I input: "readsb --device-type rtlsdr --raw --net" or "readsb --device-type rtlsdr--interactive --net", the aircraft signal can be displayed normally. 
**However, when I open the webpage: IP adress/tar1090, it shows "Problem fetching data from the server: Seems the decoder /receiver /backend isn't working correctly".** 
I tried the command: "**sudo journalctl --no-pager -u readsb**", and it showed "**FATAL: rtlsdr: error opening the RTLSDR device: Device or resource busy**" and "**readsb.service: Scheduled restart job, restart counter is at 144**". 
I also tried "_**echo -e 'blacklist rtl2832\nblacklist dvb_usb_rtl28xxu\nblacklist rtl8192cu\nblacklist rtl8xxxu\n' | sudo tee /etc/modprobe.d/blacklist-rtl-sdr.conf", but it didn't work.**_ 
Does anyone know how to solve this problem?
[image]
[image]

wiedehopf, 2024-08-09:
https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

Did  you use this?

Also check if user readsb is in plugdev:
```
groups readsb
```

For the blacklist to take effect  you need to reboot.

kevin4396, 2024-08-09:
hello!
Thank you for replying to me.
I am a newbie in readsb+tar1090. I have solved this problem. I will tell you the process, hoping to help people who have the same problem as me.
In fact, I complicated a simple problem.
I typed: "sudo journalctl --no-pager -u readsb", inserted RTLSDR, and did nothing. I&nbsp; checked the log ,and found that message like&nbsp;"FATAL: rtlsdr: error opening the RTLSDR device: Device or resource busy" and "readsb.service: Scheduled restart job, restart counter is at 144"&nbsp; disappered&nbsp;, and tried to open "IP adress/tar1090". I found that I solved the problem.
I used antirez/dump1090 before. I needed to enter "./dump1090 --interactive --net" in the terminal before I could open the web page to see the map.
However, in readsb+tar1090, the readsb service is self-starting. After I entered "readsb --device-type rtlsdr --raw --net" in the terminal, the command I entered prevented readsb from starting automatically. So I encountered this problem. In fact, I just need to insert RTLSDR and click on "IP adress/tar1090".

&nbsp;

------------------&nbsp;原始邮件&nbsp;------------------
发件人:                                                                                                                        "wiedehopf/tar1090"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2024年8月9日(星期五) 中午12:32
***@***.***&gt;;
***@***.******@***.***&gt;;
主题: &nbsp;Re: [wiedehopf/tar1090] 设备或​​资源繁忙：黑名单不起作用 (问题 #328)

 
https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb
 
你用过这个吗？
 
还要检查用户 readsb 是否在 plugdev 中：
 groups readsb  
为了使黑名单生效，您需要重新启动。
  
—
直接回复此电子邮件、在 GitHub 上查看或取消订阅。
您之所以收到此邮件，是因为您是该主题的作者。消息ID ：< ***@***.*** &gt;

closed 2024-08-09 by kevin4396