wiedehopf/tar1090 #329: How to get  history aircraft.json  like "history_0.json, history_1.json, ..., history_119.json"?

kevin4396, 2024-08-14:
Hello!
I read the readsb/README-json.md ,and know that the history aircraft.json can be obtained ,from the link:https://github.com/wiedehopf/readsb/blob/dev/README-json.md#history_0json-history_1json--history_119json

**Problem**:However, when I pluged in rtlsdr, and started readsb+tar1090,  typed "**cd /run/readsb/**", I just found  aircraft.json without files like "history_0.json, history_1.json, ..., history_119.json",**I wait for 3 minutes ,but I still did't find  history aircraft.json.**
[image]

If I want to get the history files, **is there anything I should set**?

kevin4396, 2024-08-14:
I use the default config ,like this 
[image]

wiedehopf, 2024-08-14:
You shouldn't need them, they are not very useful.

But here you go, reinstall readsb newest version and add this command line argument.
```
--devel=legacy_history
```

wiedehopf, 2024-08-14:
Oh the readme needs fixing.

Should just remove that section, no one should be using those files.

closed 2024-08-14 by wiedehopf