wiedehopf/tar1090 #338: Authentication for tar1090 site

jdschuitemaker, 2024-09-17:
I would like to be able to access my ads-b / tar1090 site on its subdomain. Not sure if this is a good or bad idea but to avoid anyone being able I was thinking if it is possible to add some kind of authentication.

Would it be possible to add https://www.auth0.com or similar service to the site?

wiedehopf, 2024-09-17:
The login you linked would not prevent anyone to just access /tar1090/data/receiver.json for example.

wiedehopf, 2024-09-17:
Just change the --lat --lon for readsb a bit so it doesn't point at your precise location.
Don't use heywhatsthat either.

Then there is no precise location..... and your only issue would be people scraping aircraft.json and using bandwidth.
You can block that easy enough in a webserver.

jdschuitemaker, 2024-09-19:
Thank you for the replies. I will look at tailscale / zerotier

jdschuitemaker, 2024-09-19:
For anyone else that wants to protect their tar1090 website that you are running on a domain or sub-domain, the below links can help if you are already using Cloudflare for your domain. Cloudflare calls this "Self Hosted Applications".

The way I have set it up now, when I open my page 'fr.domain.com' Cloudflare asks to provide an email address (registered during the setup) and then receive a pin to validate. If you don't validate, you cannot open the site and also not the data at /tar1090/data/receiver.json. If you provide a valid pin, Cloudflare will forward to your tar1090 page.

This video helped me: [https://www.youtube.com/watch?v=Ynr8VubJqvY](url)
And this page provides written documentaton from Cloudflare: https://developers.cloudflare.com/cloudflare-one/applications/configure-apps/self-hosted-apps/

wiedehopf, 2024-09-19:
reopening for visibility.
i'll close it maybe in the future.