wiedehopf/tar1090 #339: How to connect readsb to an rtlsdr device by serial number?

Arizona-Steve, 2024-10-04:
How do I configure readsb connect to an rtlsdr device by serial number?

I saw somewhere to change --device 0 to --device &lt;serial number&gt; in /etc/default/readsb
When i do that, "systemctl restart readsb" succeeds and the readsb service is active
but when I reboot, it gets changed back to --device 0

I am trying to get the UAT side of my Nooelec Flycatcher hat working.
I installed tar1090 from your script
I installed dump1090-fa from the instructions on the FlightAware piAware pages.
I changed the SN in the UAT rtlsdr device, configured /etc/default/dump1090-fa to use it
and configured /etc/default/tar1090 to enable UAT as per your instructions.

When I plug in both the ADSB and UAT USB cables to the PI, ADSB is not working.
"systemctl status ..." shows readsb service is active and dump1090-fa is not
"journalctl" shows no problems on readsb and the soapySDR device is busy on dump978-fa.
I think readsb is grabbing the UAT rtlsdr device because occasionally a very few  planes show up,
which I think are the strongest signals making it through the 978 SAW filter on the UAT side of the flycatcher.

Thank you
Steve

wiedehopf, 2024-10-04:
Either you're not saving the file or your sd-card is broken.

Post the log for readsb startup and dump978-fa startup.

You wrote dump1090-fa multiple times ... use either dump1090-fa or readsb, not both.

Arizona-Steve, 2024-10-04:
My bad, I am only using readsb and dump978-fa.
I have fixed the errors in my previous post.

I am rebuilding on a different SD card.
Where do I find the logs that you want?

Thank you,
Steve

wiedehopf, 2024-10-04:
journalctl -u readsb
journalctl -u dump978-fa

just in case you want an all in one solution, have a look at adsb.im

Arizona-Steve, 2024-10-05:
After rebuilding on a different SD card, it appears to be working.
Both readsb and dump978-fa services are active 
and the serial number I added after --device in /etc/default/readsb survives a reboot.
I am seeing both ADSB and UAT traffic.

Thank you so much
Steve

wiedehopf, 2024-10-05:
https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

closed 2024-10-05 by wiedehopf