wiedehopf/tar1090 #341: No BIAS-T

ExquisiteMetropolis, 2024-10-10:
I had my setup previously running on a different seperate docker container. That offered the option to enable BIAS-T for my RTL dongle.
Ref: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
I do prefer to have this running as HA add-in.

Since a power outage i noticed my reception dropped and BIAS-T was off (no power to filter).

When I temporary ran my old docker container to enable BIAS-T again, the reception was back to normal. 

Can you add the BIAS-T option to this container? Should be on the dump1090 part.

wiedehopf, 2024-10-10:
this is very much the wrong place for this inquiry.

https://sdr-enthusiasts.gitbook.io/ads-b
https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder

The container you're using is not actively maintained, consider the links above.
Further issues raise with the container, not this project.

closed 2024-10-10 by wiedehopf

ExquisiteMetropolis, 2024-10-10:
I think you missed the question.
I linked the other container as working example. Your add-in within HA doesn't provide a BIAS-T option. That's what I am looking for.

wiedehopf, 2024-10-10:
There is no "my HA add-in".

Which container are you running?

ExquisiteMetropolis, 2024-10-10:
Then I must have misread. This page is linked at the bottom of this one: https://community.home-assistant.io/t/home-assistant-add-on-ads-b-multi-portal-feeder-flightradar24-and-flightaware/345563

wiedehopf, 2024-10-10:
Quoting the first line from that link:
> I wrapped the awesome docker image from [Thom-x 396](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090) as Home Assistant Add-On.

Anyhow can't you just run this stack:
<https://sdr-enthusiasts.gitbook.io/ads-b>

I suppose you're looking for clicky things.
I'd recommend https://adsb.im/home but that likely won't play nice with home assistant.
But if you have an extra pi, adsb.im would be the way to go if you don't want to deal with config files.