wiedehopf/tar1090 #342: ADSB local low altitude loging

Mike-D-424, 2024-10-25:
I wish to find a magic way for my existing ADS-B feed to create and (preferably e-mail me) a daily report, including aircraft ID, speed, heading, date, time, altitude, etc., of any aircraft lower than 1000 ft that my device receives.

I want this because I am tired of helicopters making my house shake since I live under their approach to a local airport.

I am not a coding or pi expert; therefore, a more straightforward and easier-to-implement solution is preferred.

Thank you, Michael

closed 2024-10-25 by wiedehopf

RundesBalli, 2024-10-27:
It's not exactly what you want, but I'd still like to share this link: https://github.com/RundesBalli/adsbTelegramNotifier

wiedehopf, 2024-10-27:
please note ADS-B shows uncorrected pressure altitude usually which is not MSL.
even if you get the geometric altitude from ADS-B, it's WGS84 referenced, not MSL for that you need converting to EGM96.

Mike-D-424, 2024-10-27:
Thank you!

Michael Del Greco
***@***.***
56 Woodland Drive
Woodland Park, NJ 07424