wiedehopf/tar1090 #350: Disabling range rings via config.js not working

jaluebbe, 2024-12-14:
I tried to disable the range rings to appear on default via config.js and SiteCircles = false; but it didn’t work.

wiedehopf, 2024-12-14:
It does.
But browser cookie is overwriting if you toggled it before.

closed 2024-12-14 by wiedehopf

jaluebbe, 2024-12-14:
I tried it using incognito windows on safari and chrome. In both cases, the variable in the config.js seems to be ignored. The range rings appear as soon I allow the browser to provide geolocation.

wiedehopf, 2024-12-14:
F12 for dev console, check for js errors.

Also look at the config.js loading in the browser and confirm its content.

wiedehopf, 2024-12-14:
Oh geolocation ... i guess i can check that.

wiedehopf, 2024-12-14:
Using globe index with geolocation is a somewhat important detail to reproduce this issue.
It's quite a bit from the default config thus me testing / closing at first.