wiedehopf/tar1090 #351: cannot collapse filter panel after opening it

axd1967, 2025-01-01:
[image]
[image]

wiedehopf, 2025-01-01:
There is an X to close the aircraft.

Also adsbexchange has a fork, they don't use this repo exactly.

axd1967, 2025-01-01:
The 'X' deselects the a/c... then only can the panel '>' appears

but it should be tar1090's?

[image]

wiedehopf, 2025-01-01:
Well it's still a fork, no matter what their site says.

Regardless this issue is probably in this version as well.
But as you see there is a workaround.
Thus won't fix as it's a rare issue.