wiedehopf/tar1090 #352: Can't distinguish between tracks on different dates

hughfe, 2025-01-12:
At about 180 to 200 miles range on some days I can pick up QF28 from Santiago to Sydney.  With ?ptracks=168 I can see 2 similar intermittent tracks. Clicking on either one shows the last position on the later track only, and the flight details for that track.
1) It would be nice if these flights could be distinguished.
2) It might be helpful if the popup sidebar with flight details also showed the date of the track.

I can use ptracks=168 because there is very little traffic here - right now there is nothing within range, even on Flightradar there is nothing in the air within 50 miles!  To compensate I regularly pickup the flights from Christchurch to Antarctica.
[image]

wiedehopf, 2025-01-12:
track labels won't show you?

wiedehopf, 2025-01-12:
You can turn on
```
--write-globe-history /var/globe_history --heatmap 15 --json-interval 15
```

and then use /?replay
plus for specific airframes you have persistent history.

This of course requires readsb, maybe you're running dump1090-fa or something then you need to switch.

closed 2025-01-12 by wiedehopf

hughfe, 2025-01-13:
Thanks for suggesting replay & labels (is ?replay documented somewhere)?  These let me distinguish between the flights.
The more southerly track is QFA28 (westbound) on 2025-01-09, the more northerly track is QFA27 (eastbound) on 2025-01-08.
Clicking on any of the 3 track segments always gives the same info about QFA28 on the last of the segments (image VB-ZNN).
Probably adding to the confusion is that it is the same aircraft on both flights.  Clicking on the tracks can't separate the 2 flights, although as you say the information is available in other ways.  So this is probably not an important issue.
[image]
[image]
[image]

wiedehopf, 2025-01-13:
no this is not what ptracks is for.
you can do all kinds of things. tar1090 does enough.

documention mentions replay in some form somewhere.
i don't really care.

oh --write-state-only-exit probably for less writes.

documenting such features better means people will ask about how to edit /etc/default/readsb because they don't know how.
i do what i feel like for my projects because really i'm somewhat done with them, minimal maintenance :)

hughfe, 2025-01-13:
Thank you.  Understood.  It's a great bit of software, thanks!