wiedehopf/tar1090 #353: [feature] Generation of map tile set more detailled

desteny59, 2025-01-13:
Hello,

i'm in an offline network and i'd like to zoom on an airport zone. Is it possible to generate lvl15/16 tile set for only the airport zone ?

wiedehopf, 2025-01-13:
not easily no.

wiedehopf, 2025-01-13:
Generating map tiles from the around 100 GB of openstreetmap database needs a lot of memory and is a PITA.

wiedehopf, 2025-01-13:
I suppose i should look at this, they seem to ask for donations but you can load tiles fairly easily: https://openfreemap.org/

It's vector tiles though and i really like raster tiles as they load the browser less and are less of a hassle software wise.
But maybe i'll have a look.

wiedehopf, 2025-01-13:
And quickly you see there is no quick start for openlayers ... https://openfreemap.org/quick_start/
Maybe i'll try and make those tiles work regardless.

desteny59, 2025-01-13:
Thank you for all these details. I managed to make a set of tiles with maperitive. You have to be careful to have each level of detail (9 -> 10 -> 11 etc.) without forgetting one otherwise it does not work.

wiedehopf, 2025-01-13:
There is some magic happening in the tar1090 install script.
It checks the max zoom level present in the directory structure and sets offlineMapDetail variable in config.js accordingly.
This way when you zoom past the max zoom level the map doesn't disappear but becomes blurry instead.