wiedehopf/tar1090 #355: TAR1090 on Windows 11

Dizzybear59, 2025-01-27:
Hi, a total noob when it comes to this sort of thing,so - I have a Windows 11 tablet that I wish to setup as a sort of mobile adsb receiver . I have dump1090 running with an RTL-SDR v.3 dongle and would really like to have a graphical display of aircraft received. As I can't find an obvious file to run, could you help me please?

wiedehopf, 2025-01-27:
You could run virtualbox and https://adsb.im/ possibly.

Zero interest in supporting windows.
Don't think the dump1090 available for windows produces the json used for tar1090.
Even if it is, you'd have to figure out the webserver part and everything on your own.

I've tried virtualbox on a windows desktop and it works fine.
Just needs USB3 passthrough configured.

Dizzybear59, 2025-01-27:
Ok, thanks for getting back to me - I'll have to look elsewhere.

closed 2025-01-27 by Dizzybear59

gvanem, 2025-02-04:
> Don't think the dump1090 available for windows produces the json used for tar1090.

My [dump1090](https://github.com/gvanem/Dump1090) version does.