wiedehopf/tar1090 #366: ACARSHub integration

tyzbit, 2025-04-26:
ACARSHub has a JSON feed for ACARS (and another for VDLM2) messages. Tar1090 could take advantage of this by showing messages received from aircraft in the UI, either as a "unread" bubble on the aircraft itself or in the side panel when viewing a particular aircraft.

I'm mostly a Go programmer but I know Javascript as well - if this seems like a useful addition I can take a stab at adding it but some general guidance on how to implement it would be very helpful as I'm unfamiliar with Tar1090s codebase.

wiedehopf, 2025-04-26:
I'm not aware it has an API via http?

Otherwise you'd have to change the backend for tar1090 which i'm not prepared to do at this point.

tyzbit, 2025-04-26:
It doesn't have an API, but it streams JSON blobs of messages. Here's an example payload:
```json
{
  "freq": 130.025,
  "channel": 0,
  "error": 0,
  "level": -33.7,
  "timestamp": 1745700345.848858,
  "app": {
    "name": "acarsdec",
    "ver": "4.1",
    "proxied": true,
    "proxied_by": "acars_router",
    "acars_router_version": "1.3.1",
    "acars_router_uuid": "72d5b013-6a4d-45be-aa00-98fdbdfc2d63"
  },
  "station_id": "STATION",
  "assstat": "skipped",
  "mode": "2",
  "label": "_d",
  "block_id": "J",
  "ack": "5",
  "tail": "N234FR"
}

```
I made a best guess at [a schema](https://github.com/tyzbit/acars-processor/blob/main/annotator_acars.go#L32) for those messages in an app I made called acars-processor. There's another, different one for VDLM2 messages as well.

My theory was to connect to and subscribe to an ACARSHub JSON port, receive JSON messages and then associate them with any aircraft and show the info in the message in a dialog. I don't know if it'd be websockets but the ACARSHub JSON port isn't an API, just a TCP/UDP port that sends JSON when it gets a message to connected clients.

wiedehopf, 2025-04-26:
Connecting TCP from js is similarly problematic.
Needs disconnecting when the tab is in the background, then it doesn't update as expected.

Also subscribe means websocket.

wiedehopf, 2025-04-26:
Make a backend (in acarshub or independent) that can be queried for all acars messages for the last X minutes.

On the js getting back to the foreground + on site load the last 30 min of acars messages (or however much) can be loaded.
After that every 15 seconds, all acars messages from the last 20 seconds are loaded.

I'm still not a fan.
tar1090 is already chaos as it is :)

But websockets need an extra js library and really i don't want to add more libraries.
That's a bit silly. At the same time i don't want to deal with non-http connections.
Maybe i just don't care for it and there are no good reasons not to do it, in that case i'm sorry! :)