wiedehopf/tar1090 #371: Airplane icons on live map are black rectangles

JasenkoC, 2025-05-23:
Hello,

I have a brand new installation of tar1090 with readsb on latest Raspbian OS. Hardware is Raspberry Pi 5 with 8GB RAM. I have also installed graph1090 and it all works fine apart from the icons of planes being black rectangles. Web server is NginX and the include statements for tar1090 and graph1090 have been added to the config file. I have tried opening the tar1090 URL in couple of different browsers but it behaves the same. I have decent sysadmin knowledge, but web apps are beyond my abilities to debug.
In the end it's not a big issue since it's just a cosmetic flaw on my setup, but it's kinda bugging me.
I should also mention that I also have Raspberry NOAA v2 installed and running alongside tar1090. It all works on two dedicated RTLSDR dongles without issues.
Thank you in advance for your help.

wiedehopf, 2025-05-23:
Different browsers but different devices too?

Probably webGL issue you can disable it, will just be slower.

JasenkoC, 2025-05-23:
I tried it just now. I was primarily browsing it from my Windows machine with Firefox, but it behaves the same on an iPhone (Safari) and on Firefox/Chromium on Fedora Linux 42 machine. I did try disabling WebGL on all devices and it just makes the planes vanish altogether. No activity is shown on the live map if I disable WebGL. I also tried loading a couple of demo web pages for WebGL and they work without any visual glitches or errors.
Anything else I could check or do?

wiedehopf, 2025-05-23:
Disable webgl via the website menu accessed via the gear wheel.

Anyhow if it's multiple devices ... it's possible that it's a bad install of tar1090.
So just rerun the installer.

You can always check the dev console for errors.

JasenkoC, 2025-05-23:
I found the "gear" icon and disabled WebGL there and now it's rendering the planes. Funny thing is that the gear is not drawn. It's just showing an empty square. Also, I've checked the browser console and I don't have any errors for WebGL, just bunch of warnings:

WebGL warning: tex(Sub)Image[23]D: Resource has no data (yet?). Uploading zeros.
WebGL warning: drawElementsInstanced: TEXTURE_2D at unit 0 is incomplete: The dimensions of `level_base` are not all positive.
WebGL warning: drawElementsInstanced: TEXTURE_2D at unit 0 is incomplete: The dimensions of `level_base` are not all positive.
WebGL warning: drawElementsInstanced: TEXTURE_2D at unit 0 is incomplete: The dimensions of `level_base` are not all positive.
WebGL warning: drawElementsInstanced: TEXTURE_2D at unit 0 is incomplete: The dimensions of `level_base` are not all positive. 
And the last one is repeating endlessly.

wiedehopf, 2025-05-23:
Not loading the sprites png?

wiedehopf, 2025-05-23:
You can also try adsb.lol, see if that works.
If it doesn't you have an issue with your device where you're viewing the site from.

JasenkoC, 2025-05-23:
I finally checked the error.log for NginX and found this:

2025/05/23 15:34:52 [error] 9882#9882: *48 open() "/srv/images/sprites_f0c84e57a9cf2e424887b81bcd5b8a7f.png" failed (2: No such file or directory), client: 192.168.1.171, server: raspberry-noaa.localdomain, request: "GET /tar1090/images/sprites_f0c84e57a9cf2e424887b81bcd5b8a7f.png HTTP/1.1", host: "weather-pi.local", referrer: "http://weather-pi.local/tar1090/?reset"

JasenkoC, 2025-05-23:
I've fixed it... I realized that something was rewriting the /images/ to /srv/images/ and I've found the default config statements in the /etc/nginx/sites-available/default file:

```
  location ~ /images/(.*) {
    alias /srv/images/$1;
  }
```

I commented it out and now it all works.
I'm pretty sure that it's the default NginX configuration that conflicts with the tar1090 URL for images and rewrites it.

wiedehopf, 2025-05-23:
That is certainly not a default configuration.

JasenkoC, 2025-05-23:
You're right. I just checked and it is a part of the Raspberry NOAA v2 configuration for NginX. I opted to install lighttpd and changing the port to 8080. I reinstalled tar1090 and graphs1090 and now they all can coexist without stepping on each other. Thanks again!