wiedehopf/tar1090 #372: [Feature] Filter by route

Sovxx, 2025-05-28:
Thanks the great work!

Would it be possible to add route filter ? Like "contains CDG".
Or, but more complicated, propose detailled filter with Departure / (via ?) / Arrival / Both, similarly to FR24 ?

Thanks

nikallian, 2025-07-10:
Is it possible to add another request on top of that ?

I also use routes (through planefinder) and I also enabled ICAO codes. Is it possible to have somehow the names of Cities/Airports by hovering the mouse over routes/ICAO/IATA codes ?

Thanks!

MariaMerkel, 2025-09-03:
@nikallian I wanted this feature on my own tar1090 instance, so I created a patch for it: https://github.com/MariaMerkel/tar1090/commit/3a80cae51f21389504c5f9ca9938c94ae22c2a6a

You could either apply this to your own fork of tar1090 or modify the deploy script to install from that repo and branch. I would however recommend against the latter option as I probably won't keep that branch up to date with the mainline tar1090 as I run a different fork for my own instance.

If @wiedehopf approves, I can  submit a pull request to his repo so this can become a feature of the standard tar1090 software.

wiedehopf, 2025-09-03:
Merged the tooltip thing.

Nothing to do with the route filtering though.

wiedehopf, 2025-09-03:
Was gonna say that i can't reproduce this.

Let's just ... always make the route column html when it is used.
Not sure how that would lead to the above but try updating, the change shouldn't hurt anything (92a477fefa9c6e4ce90663f6a1802e31821f5407)

closed 2025-09-03 by wiedehopf