wiedehopf/tar1090 #379: [Feature] Add Hybrid Satellite Imagery

Nekzuris, 2025-07-11:
For this usage I think satellite imagery alone isn't enough, it would be much more useful with labels and borders.
ADS-B Exchange premium as one.
[image]

Nekzuris, 2025-07-11:
care to elaborate? just time or other reasons?

wiedehopf, 2025-07-11:
You didn't link a map tile source that is free.

You can guess why that layer is only in premium if you want.

wiedehopf, 2025-07-11:
Oh they do that as a vector tile layer, that's a PITA to implement.
At least if that's this: https://www.arcgis.com/home/item.html?id=86265e5a4bbb4187a59719cf134e0018

I'll reopen the issue so people who want to deal with this can deal with it, I won't.
Maybe someone else has an idea.