wiedehopf/tar1090 #384: Surface broken since last update

philipp-zm, 2025-08-24:
Hello! 

I just installed the latest update. Since then, the interface has stopped working. I've tried everything I can think of, but nothing works.

Error description: Update installed without error messages. Everything was working perfectly before the update. I've never had any problems before. Now, since the current update, the map remains black. Only when I deactivate WebGL in the interface can I see the map. But even then, no aircraft are displayed on the map. They are invisible. When I move the mouse over the empty map, the flags of the aircraft currently in flight are displayed, but nothing can be seen on the map itself. 

I would appreciate some help. If you need more information about my system, please feel free to contact me.

Thanks in advance!

[image]
[image]

wiedehopf, 2025-08-24:
You have to disable webGL in the settings via the settings gear icon.
That will enable the non-webGL aircraft icons.

webGL not working is very likely an issue with your browser / system.
There is attempts made to detect it automatically and fall back to non webGL but lots of systems are so weirdly broken that it doesn't even error properly.

philipp-zm, 2025-08-24:
> You have to disable webGL in the settings via the settings gear icon. That will enable the non-webGL aircraft icons.
> 
> webGL not working is very likely an issue with your browser / system. There is attempts made to detect it automatically and fall back to non webGL but lots of systems are so weirdly broken that it doesn't even error properly.
> 
> I doubt i can help you more than telling you how to disable webGL thus if that works for you, please close.

Thanks for your help! That's how it works. I didn't realize there was another WebGL switch. Thanks to your help, I found it and disabled it. Now I can see the aircraft again. I still don't understand why WebGL suddenly stopped working, because it was working fine before the update. But the problem is solved for now. Thanks for that! And thank you very much for providing your great project! All the best! Bye!

[image]

closed 2025-08-24 by philipp-zm